JVital2013/vitality-goes #9: Port 8125 in use Graphite load

Dentonknifeworks, 2022-06-19:
When I start or try to load Graphite I get an error p 8125:8125 is in use.

I have Grafana on it that I use to monitor it.
I could remove Grafana but Id rather not.
So I am trying to figure out what port list should I move this to and will it work?
The nice-looking interface still working on configuring but it's working.

Thanks, Brad

creinemann, 2022-06-19:
Brad, I tried multiple ways as well, with no success, editing the TCP.ini. and UDOP.ini, as well as changing the port assignments in goesrecv.conf files.  No Joy either. Works great on my test pi with no grafana.  Simple fix would be to alter the script.js to point to the grafana address and bring that up instead of graphite until a fiz is found.

Dentonknifeworks, 2022-06-19:
Well I had a look at script.js and nope can't not even sure where to change that LOL

JVital2013, 2022-06-19:
It appears that goesrecv only lets you report statistics to a single statsd host, so right now you can either use Grafana or Graphite. I'll see what it takes to give Vitality GOES users the choice between using graphite or grafana as the statistics database

Dentonknifeworks, 2022-06-19:
Well, there it is the answer. I will be looking at how I can add Grafana in it since I have it set up anyway.
Thanks
Brad

JVital2013, 2022-06-20:
I'm digging into this, and I'm going to change tactics.

While Grafana does have an API that can be used to pull graphs, it's seriously different from graphite. Vitality GOES would need two separate statistics handers, which sounds like unnecessary complexity to me. On top of that, grafana "panels" do not have deterministic names in the API. There would need to be a complex config on the Vitality GOES side so it even knows where to pull the panels - something that can be done with one line.

This is a lot of complexity just so users can use secondary software. In an attempt to avoid more complexity, graphite is going to stay the the only source of graphs in Vitality GOES.

That being said, not all hope is lost. There are solutions, like those posed [here](https://unix.stackexchange.com/questions/532029/iptables-wrong-destination-ip-of-duplicated-udp-packet), that would allow users to duplicate the statsd packets from goesrecv to both graphite and grafana. I'll do some testing to see if I can get this working reliably. If I can get it working well, I'll do a writeup on the best way to configure it.

For now, if you're affected by this bug, just comment out the graphiteAPI line in Vitality GOES's config.ini with a semicolon (;). The software will keep working, you just won't get the graphs.

JVital2013, 2022-06-20:
Potentially simpler userland service which could do this as well (leaving here for my notes, will not work as-is)

```
socat -U - udp4-recv:8325 | tee >(socat -u - udp4-datagram:127.0.0.1:8125) | socat -u - udp4-datagram:127.0.0.1:8225
```

JVital2013, 2022-06-24:
For those of you that are having this issue, try out the fix in the pull request - specifically, it's documented [here](https://github.com/JVital2013/vitality-goes/blob/dc2265aded8bdf90dbafb3f13481db41f8c74513/docs/grafana-compatibility.md).

The fix consists of two pieces:

1. Graphite would not start on some versions of Raspberry Pi OS due to an old version of libseccomp2. **This is what caused the 502 bad gateway error.** Work around it by using the `--privileged` flag in the `docker run` command
2. I'm include a simple "service" that accepts statsd packets from goesrecv on port 8325. It will then forward the packets to both Graphite (on port 8225) and Grafana (on port 8125). **This prevents the 8125 port conflict.**

Note these changes might require tweaks to what you already have. It should all be outlined in the [doc linked above, but linked here again just for fun.](https://github.com/JVital2013/vitality-goes/blob/dc2265aded8bdf90dbafb3f13481db41f8c74513/docs/grafana-compatibility.md)

Dentonknifeworks, 2022-06-25:
Ok Now Graphite loads and Grafana is working and showing data but Graphite is not showing data?

JVital2013, 2022-06-25:
@Dentonknifeworks Let's open a separate issue if Graphite isn't receivind data since that's a separate issue behind-the-scenes. When you open a new issue, I'll need to know:

- Are graphite, grafana, and goesrecv all on the same system, or if they're on different systems, what are their local IPs?
- Is statsdduplicator running? (What's the output of `systemctl status statsdduplicator`)
- What does the `monitor` section of your goesrecv.conf look like?

PR #17 (2022-06-25): StatsD duplicator and Graph Fix
This pull contains 2 fixes:

1. **Graphite would not install on some versions of Raspberry Pi OS due to an old version of libseccomp2.** Work around it by using the --privileged flag
2. **It appears goesrecv cannot send statistics to multiple statsd receivers at once.** So, if you have Grafana set up, how can you keep using it? To work around the issue, I'm include a simple "service" that accepts statsd packets from goesrecv. It will then forward the packets to both Graphite and Grafana.

Fixes #9
closed 2022-06-25 by JVital2013