JVital2013/vitality-goes #13: Video rendering does not support dated subdirectories

JVital2013, 2022-06-20:
The included video rendering scripts do not support having images in subdirectories. This needs fixed

jerry110, 2022-07-16:
Getting the Same.....
`[2022-07-15 23:26:16] Creating GOES16FalseColor...
cp: cannot stat '/SatData/goes16/fd/fc/*_20220708*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x5613b1b02300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-15 23:26:16] Creating GOES16FalseColorNoBorders...
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220708*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x558cb8d62300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-15 23:26:17] Creating GOES16Ch2...
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220708*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x55ba77a10300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-15 23:26:17] Creating GOES16Ch7...
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220708*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x5649f1968300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-15 23:26:18] Creating GOES16Ch7Enhanced...
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220708*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch07_enhanced/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x55b1450d4300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-15 23:26:18] Creating GOES16Ch8...
cp: cannot stat '/SatData/goes16/fd/ch08/*_20220708*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch08/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch08/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch08/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch08/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch08/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch08/*_20220714*.jpg': No such file or directory
[... cut ...]
            ├── GOES16_M2_FC_20220716T010755Z.jpg
            ├── GOES16_M2_FC_20220716T012255Z.jpg
            ├── GOES16_M2_FC_20220716T013755Z.jpg
            ├── GOES16_M2_FC_20220716T015255Z.jpg
            ├── GOES16_M2_FC_20220716T020755Z.jpg
            ├── GOES16_M2_FC_20220716T022255Z.jpg
            ├── GOES16_M2_FC_20220716T023755Z.jpg
            ├── GOES16_M2_FC_20220716T025255Z.jpg
            ├── GOES16_M2_FC_20220716T030755Z.jpg
            └── GOES16_M2_FC_20220716T032255Z.jpg

57 directories, 228 files
`

jerry110, 2022-07-16:
Now using vitality-goes [goesproc-goesr.conf](https://github.com/JVital2013/vitality-goes/blob/main/goestools-conf/goesproc-goesr.conf)
[goesrecv.conf](https://github.com/JVital2013/vitality-goes/blob/main/goestools-conf/goesrecv.conf)

And here is the output of CreateVideos-ABI.sh
`[2022-07-16 16:52:20] Creating GOES16FalseColor...
cp: cannot stat '/SatData/goes16/fd/fc/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x562edc2ff300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-16 16:52:21] Creating GOES16FalseColorNoBorders...
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220715*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220716*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x5615f8a74300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-16 16:52:21] Creating GOES16Ch2...
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220715*.jpg': No such file or directory
^C
root@Goes:/SatData/scripts# nano CreateVideos-ABI.sh
root@Goes:/SatData/scripts# ./CreateVideos-ABI.sh
[2022-07-16 16:53:14] Creating GOES16FalseColor...
cp: cannot stat '/SatData/goes16/fd/fc/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x55dbfc698300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-16 16:53:15] Creating GOES16FalseColorNoBorders...
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220715*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/fc-noborder/*_20220716*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[image2 @ 0x55c2c4da4300] Could not open file : *.jpg
Output file #0 does not contain any stream
[2022-07-16 16:53:16] Creating GOES16Ch2...
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220709*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220710*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220711*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220712*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220713*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220714*.jpg': No such file or directory
cp: cannot stat '/SatData/goes16/fd/ch02/*_20220715*.jpg': No such file or directory
mogrify-im6.q16: unable to open image `*.jpg': No such file or directory @ error/blob.c/OpenBlob/2924.
[... 1477 lines cut ...]
[... cut ...]
            ├── GOES16_FD_TPW_20220716T095021Z.jpg
            ├── GOES16_FD_TPW_20220716T105021Z.jpg
            ├── GOES16_FD_TPW_20220716T115021Z.jpg
            ├── GOES16_FD_TPW_20220716T125021Z.jpg
            ├── GOES16_FD_TPW_20220716T135021Z.jpg
            ├── GOES16_FD_TPW_20220716T141021Z.jpg
            ├── GOES16_FD_TPW_20220716T165021Z.jpg
            ├── GOES16_FD_TPW_20220716T175020Z.jpg
            ├── GOES16_FD_TPW_20220716T185020Z.jpg
            └── GOES16_FD_TPW_20220716T195020Z.jpg

39 directories, 1044 files
`

JVital2013, 2022-07-16:
@jerry110 your new issue is because you only have ABI images from today. The ABI video script only renders timelapses from 7 days ago to midnight “satellite time” this morning. It does this to make the fc images loop nicer. Since your images folder only has images from today, nothing is rendered. The EMWIN script will render all images into a video, however.

Since your second comment is long and not related to this issue, I'm going to hide them. We can continue discussion on Facebook

JVital2013, 2022-07-18:
The script rewrite is complete and is available in the "Script Overhaul" pull request above. This issue is fixed in the changes. I'm going to run them on my system for a day or so, then merge the pull request and push a new release.

Feel free to try it now and let me know how it works for you

PR #24 (2022-07-19): Script overhaul
Reworks some of the scripts to solve the following problems:

- **Bug Fix:** Fixes #13. It is typical to have dated subdirectories when saving images with goesproc. Previously, the scripts only worked on "flat" goesproc directory structures - i.e. ones without dated subdirectories.
- **Improvement:** Moved script configuration to `scriptconfig.ini`. If there are future updates to the scripts, it should be as simple as dropping the new scripts in place. No more need to re-edit the script header every update!
- **Performance:** Remove unnecessary file copy in `CreateVideos-EMWIN. [... cut]
closed 2022-07-19 by JVital2013