JVital2013/vitality-goes #15: Issues using SD card.

pumatrax, 2022-06-23:
Hi, 

I am having a problem using a sdcard with vitality. For a while I have been using goestools to save to a sdcard connected via usb but I just added Vitality and used your goesproc config file and edited the locations like below. The files are saving but pointing the config.ini to the sdcard location throws a lot of errors in the apache log and nothing is visible in the web ui. In the apache error.log I see errors pertaining to 'permission denied' when its trying to access the usb sdcard?

I also did not see the configuration option, current weather, other emwin when it was set to pull data from sdcard. I have relocated the data to the pi sdcard it boots from and its  all now working.

Any suggestions would be appreciated. 

# EMWIN
[[handler]]
type = "emwin"
directory = "/media/pi/SDCARD/emwin"

JVital2013, 2022-06-23:
Sounds like the www-user account does not have permission to access /media/pi/SDCARD. What are the permissions on /media/pi/SDCARD?

`ls -l /media/pi/` should get that for you.

pumatrax, 2022-06-23:
Hi,

pi@hampi:~/Desktop $ ls -l /media/pi/ 
total 16
drwxr-xr-x 7 pi pi 16384 Dec 31  1969 SDCARD

JVital2013, 2022-06-23:
Interesting, those permissions show that everyone on your pi should have access to the /media/pi/SDCARD directory (the last r-x shows that).

Could you run `ls -l /media/pi/SDCARD/emwin/` and see what the permissions are directly on those files? I don't need to know the permissions to all the files in the directory, just some of them. Most of the permissions are probably the same.

If you're able to attach the apache error.log file, that might shed some light on it as well

pumatrax, 2022-06-23:
Hi,

They all seem to be the same. The scripts seem to fail trying to access that drive as well. Very odd, goesproc writes to it with no problem. 

`-rw-r--r-- 1 pi pi   1445 Jun 23 03:57 A_WWUS86KHNX230757_C_KWIN_20220623075730_206394-1-SPSHNXCA.TXT
-rw-r--r-- 1 pi pi   1219 Jun 23 06:04 A_WWUS86KHNX231004_C_KWIN_20220623100412_211521-1-SPSHNXCA.TXT
-rw-r--r-- 1 pi pi   1507 Jun 22 20:22 A_WWUS86KMTR230022_C_KWIN_20220623002256_191086-1-SPSMTRCA.TXT
-rw-r--r-- 1 pi pi   1775 Jun 22 20:44 A_WWUS86KMTR230043_C_KWIN_20220623004412_191867-1-SPSMTRCA.TXT
-rw-r--r-- 1 pi pi   1555 Jun 22 23:54 A_WWUS86KMTR230354_C_KWIN_20220623035431_198154-1-SPSMTRCA.TXT
-rw-r--r-- 1 pi pi   1522 Jun 23 03:51 A_WWUS86KMTR230750_C_KWIN_20220623075110_206194-1-SPSMTRCA.TXT
-rw-r--r-- 1 pi pi  68739 Jun 22 23:49 Z_EIIO00KWBC230345_C_KWIN_20220623034538_197940-4-INDCIRUS.JPG
-rw-r--r-- 1 pi pi  70310 Jun 23 02:50 Z_EIIO00KWBC230645_C_KWIN_20220623064534_203725-4-INDCIRUS.JPG
-rw-r--r-- 1 pi pi 234220 Jun 22 19:30 Z_EINA00KWBC221130_C_KWIN_20220622233033_188641-4-G16CIRUS.JPG
-rw-r--r-- 1 pi pi 232990 Jun 22 19:46 Z_EINA00KWBC221145_C_KWIN_20220622234532_189238-4-G16CIRUS.JPG
-rw-r--r-- 1 pi pi 225564 Jun 22 21:01 Z_EINA00KWBC230100_C_KWIN_20220623010040_192442-4-G16CIRUS.JPG
-rw-r--r-- 1 pi pi 224048 Jun 22 21:16 Z_EINA00KWBC230115_C_KWIN_20220623011537_193107-4-G16CIRUS.JPG
`

JVital2013, 2022-06-23:
Ah, I'm pretty sure I know what it is.

Have you configured your fstab to mount the drive to /media/pi/SDCARD, or are you letting it auto-mount there? If you're letting it auto-mount, that's the problem. You're being affected with this issue: [https://raspberrypi.stackexchange.com/questions/105836/permission-denied-to-access-media-pi-folder](https://raspberrypi.stackexchange.com/questions/105836/permission-denied-to-access-media-pi-folder).

To quote the correct answer on that question: "This happens because you have mount helpers which mount the storage devices with the identity of the user holding the current session on your machine. If you have a GUI and log in as pi, that's the user who will own the mounted devices."

Goesproc can save the files no problem since that process is running as pi, but Vitality uses Apache, which runs as the www-data user. To solve the problem, you'll want to specify the external device's mount point in your fstab.

Again to quote from the linked answer above (but modifying for this case), add a line line this to your /etc/fstab/:

```
UUID=0123456789ABCDE   /media/SDCARD   ext4   defaults,uid=www-data,gid=www-data
```

You can view the UUIDs of your drive volumes using blkid, and you may need to replace ext4 with the file system you're using on the SD card. Also note I took it out of the pi subdirectory so it doesn't conflict with the auto mounter.

pumatrax, 2022-06-23:
The apache error.log grows wildly out of control when I change the location for emwin but here is a little bit of the log. You see the permission error once and then the log fills up to 100'mb in seconds. 

[Thu Jun 23 16:27:50.826009 2022] [mpm_prefork:notice] [pid 31619] AH00163: Apache/2.4.53 (Raspbian) configured -- resuming normal operations
[Thu Jun 23 16:27:50.826995 2022] [core:notice] [pid 31619] AH00094: Command line: '/usr/sbin/apache2'
[Thu Jun 23 16:29:19.942882 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  opendir(/media/pi/SDCARD/emwin): failed to open dir: Permission denied in /var/www/html/functions.php on line 34, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943033 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943080 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943112 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943141 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943170 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943198 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943226 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943255 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943283 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943314 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943342 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943371 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/
[Thu Jun 23 16:29:19.943400 2022] [php7:warn] [pid 31623] [client 192.168.2.33:58564] PHP Warning:  readdir() expects parameter 1 to be resource, bool given in /var/www/html/functions.php on line 35, referer: http://hampi.local/

pumatrax, 2022-06-23:
Hi, I did it and vitality can write to it now, but I cant copy anything to it, even though I added the pi user to www-data group. The configuration is also a little wacky not showing drop down options.

closed 2022-06-23 by JVital2013

pumatrax, 2022-06-23:
It's all working, thank you! We can close this issue. I am now troubleshooting the create videos scripts. I will open another issue for it.