JVital2013/vitality-goes #18: Graphite data not loading

Dentonknifeworks, 2022-06-25:
Graphite and Grafana on the same PI
Local IP is 192.168.254.35
monitor section of my goesrecv.conf    
[monitor]
statsd_address = "udp4://192.168.254.35:8125"

statsdduplicator.service - StatsD Statistics Duplicator
   Loaded: loaded (/lib/systemd/system/statsdduplicator.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2022-06-25 11:20:00 CDT; 15s ago
     Docs: https://www.github.com/JVital2013/vitality-goes/docs/grafana-compatibility.md
  Process: 26586 ExecStart=/bin/bash -c socat -U - udp4-recv:8325 | tee >(socat -u - udp4-datagram:127.0.0.1:8225) | socat -u - udp4-datagram:127.0.0.1:8
 Main PID: 26586 (code=exited, status=127)

Got to be a simple fix I think

Thanks Brad

JVital2013, 2022-06-25:
Yeah I think it's something simple. A couple things:

- If you're using statsdduplicator, change the port in goesrecv.conf to 8325. If goesrecv is on the same system as Graphite and Grafana, it's preferable for that line to be set to `statsd_address = "udp4://127.0.0.1:8325"`. Making this change will brake stats to Grafana as well until you follow up on this second point...
- It looks like statsdduplicator.service isn't running. Is `socat` installed on your system? You can just try running socat in a shell. If not, try `sudo apt install socat`. Then, do a `systemctl restart statsdduplicator`.

I assumed socat part of the OS, but maybe not in all cases. If it's not included as a part of the OS, I'll add it as a prerequisite install.

Dentonknifeworks, 2022-06-25:
Yes I noticed that I needed to change the Monitor port   DONE
socat was not a part of my system. INSTALLED
Re did your fix and now have data.
Looks like it was the lack of socat on the OS

JVital2013, 2022-06-25:
Excellent thanks! I'll update the docs to reflect this

Dentonknifeworks, 2022-06-25:
Yes that would be good closing the issue THANS

commit d00f672f (2022-06-25): Merge pull request #19 from JVital2013/socat-install

Require socat (Fixes #18)
closed 2022-06-25 by Dentonknifeworks