JVital2013/vitality-goes #30: Correct goesproc-goesr.conf to reflect Himawari-9 as prime JMA satellite

creinemann, 2022-12-20:
A simple updating to the goesproc-goesr.conf to make goestools correctly list the satellite name as Himawari-9 (since Himawari-8 is no longer prime)

`# Images relayed from Himawari-9.
[[handler]]
type = "image"
origin = "himawari8"
directory = "/path/to/goestoolsrepo/himawari9/{region:short|lower}"
filename = "Himawari9_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/local/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"
`

JVital2013, 2022-12-21:
Will do! I'll update the references a few other places as well

JVital2013, 2022-12-21:
Fixed in e2402ee.

I also merged the SatDump compatibility branch since it's required will make Himarari imagery viewable, even when received with goestools. The "standard" goesproc config floating around does not separate channels into its own subdirectory.

The GOES-17 and GOES-18 sample configs have been updated to split out Himawari channels without separating their folders.

closed 2022-12-21 by JVital2013