JVital2013/vitality-goes #39: GK-2A Data

KiwiInNZ, 2023-01-19:
Hi James

Understand you're looking to get access to GK-2A data. I receive the LRIT data direct from the satellite. Would that be useful?

JVital2013, 2023-01-19:
Yes it would! Ideally I'd like to get a 1 hour (or longer) IQ or cadu recording from it.

KiwiInNZ, 2023-01-21:
Hi Jamie

I can do either or both.

https://kiwiweather.com/downloads/GK-2A-20230113.zip - this has 24 hours worth of downloads with processed images / data files. Let me know when you've downloaded it so I can delete it. The file is 310MB.

For an IQ, the easiest way is for me to capture data using rtl_fm and to pipe it to a .wav file. Please provide the exact command(s) you'd like me to run and I can do this as a capture and upload it to my website, as per the above .zip file.

Regards Mike

JVital2013, 2023-01-21:
I downloaded that file, thanks! Even if I get nothing else from GK-2A, this would suffice. Which program did you get this with? Are you using goesrecv with xrit-rx?

If so, I could walk you through how to get an IQ fairly easily without completely stopping your reception pipeline

KiwiInNZ, 2023-01-21:
Yes, goesrecv  + xrit-rx.

And being able to capture what you want without impacting image capture is even better.

JVital2013, 2023-01-21:
I get it! You'll likely need to restart the services, but that's only a second of impacted capture. First, get your system ready by following these steps:

1. Look in your goesrecv.conf file to see if there's a `[rtlsdr.sample_publisher]` section. There probably isn't, so set one up like this to expose the RTLSDR sample publisher on port 5000:
    ```ini
    [rtlsdr.sample_publisher]
    bind = "tcp://0.0.0.0:5000"
    send_buffer = 2097152
    ```
2. Restart goesrecv/xrit-rx if you needed to do step 1
3. Install nanomsg-utils, if it's not already installed: `sudo apt install nanomsg-utils`

Finally, run the following command to capture a .s8 IQ by running the following command:
```bash
nanocat --sub --connect tcp://127.0.0.1:5000 --raw > gk2a-baseband.s8
```

I will need to know the sample rate configured in goesrecv.conf. The saved file will be quite large depending on the sample rate  - up to 20GB/hr! I'd like an hour, but I can accommodate whatever is reasonable to you. If you have some kind of crazy upload limit I'll take more but I'm not expecting that 😃. If downloading from your site works that's great - otherwise we can arrange some kind of SFTP transfer.

From what I can tell, there's no harm in leaving the `[rtlsdr.sample_publisher]` section in goesrecv.conf when you're done. It lets you run other tools as well, like [goesbetween](https://github.com/JVital2013/goesbetween).

Thanks for your willingness to help!

KiwiInNZ, 2023-01-22:
I've kicked off some data collection, which will be limited by the spare space on the SD card for my Pi using the instructions provided (nice and easy!).

From goesrecv.conf:
sample_rate = 1024000
Will try to get close to an hour(ish).

As before, will post a link to download it from my website. Let me know when you've got it so I can delete. It is running off a Gigabit fibre connection, although only about 500Mbps upload), so should be reasonable for download performance at your end.

KiwiInNZ, 2023-01-22:
https://kiwiweather.com/downloads/gk2a-baseband.s8
3747610624 bytes, so about 3.7GB, covering just over an hour.
Let me know when I can delete it plus how long it took to download!

JVital2013, 2023-01-22:
I pulled the file; thanks! It took 16 minutes, 2 seconds to download. If you get that throughput to your Pi, you could stream it live for decoding around the world if you're bored some day :smile:.

I'll decode the baseband later and start implementing it into Vitality GOES!

[image]

KiwiInNZ, 2023-01-22:
Sounds like you're on the right path to add GK-2A functionality. Shout out where I can help further.

PR #40 (2023-01-27): Version 1.4
## Sample Configs have been added for the following satellites:

- EWS-G1 (GOES-13) GVAR - Thanks @ZbychuButItWasTaken! Hope you have a beefy computer to view 'em
- GEO-KOMPSAT 2A (GK-2a) LRIT. Fixes #39 - Thanks @KiwiInNZ!

## Changes in this release:

- **See the "Breaking Change" section below before updating!** The imagery categories in the main menu are now customizable! Instead of being stuck with "Full Disk", "Level 2 Imagery", "EMWIN Imagery", and "Mesoscale Imagery", you can now change the name, icon, and contents of all these pages - and add your own categories!
- On GOES [... cut]
closed 2023-01-27 by JVital2013