JVital2013/vitality-goes #44: Graphite charts not working.

pumatrax, 2023-07-17:
[image]

Hi, I recently performed an update on my raspberry pi and I am now noticing I am not seeing the history in the graphite charts.

JVital2013, 2023-07-18:
It looks like graphite isn't getting the statsd data. Did your goesrecv config change?

pumatrax, 2023-07-19:
[image]

I noticed it stopped working after a raspberry software update. I am seeing this chart failing. 

For what is worth I also get a server returned error 500 when trying to load the Other-EMWIN.

I ended up reloading your software and set everything up again, same result.

JVital2013, 2023-07-19:
Let's do some debugging:

1. Please post your `/var/www/html/config/config.ini` file to help make sense of later errors
2. Set `debug = true` in config.ini
3. Go to `http://your-goes-ip/dataHandler.php?type=metadata&id=otherEmwin` and let me know what the error message says

pumatrax, 2023-07-19:
Hi, 

The graphite system does seem to be working. I can go to the graphite site on 8080 and I see the charting there. 

Here is my config.ini.
'
[general]
#siteTitle = "My GOES-16 Data"
#siteTheme = "light"
graphiteAPI	= http://127.0.0.1:8080/render/
#adminPath = /media/SDCARD/text
emwinPath = /media/SDCARD/emwin
fastEmwin = false
spaceWeatherAlerts = false
showSysInfo = true
debug = false

[paths]
GOES16 = /media/SDCARD

[categories]
abi = abi.ini
meso = meso.ini
l2 = l2.ini
nws = nws.ini
emwin = emwin.ini

[location]
radarCode = NTHES
stateAbbr = NY
wxZone = NYZ074
orig = OKXNY
rwrOrig = OKXNY
city = Staten Island
lat = xxxxx
lon = -xxxxxx
timezone = America/New_York

[otheremwin]
ini = otheremwin.ini
allowUserLoader = true
maxUserFiles = 1000
'

I resolved the 500 error issue. I commented out adminPath. I don't seem to have files in that folder.

pumatrax, 2023-07-19:
[image]

Just to be clear, I dont see much in graphite, just some areas I can see graph data.

JVital2013, 2023-07-19:
Thanks - all that looks good. Looks like data is making it to graphite. Technically, comments in ini files should start with a semi-colon (; - thanks PHP), but I don't think that's causing your issue.

With debugging still on, can you send the output of http://your-goes-ip/dataHandler.php?type=metadata&id=viterbiContent? If the output is long, put it in as an attachment or pastebin.

pumatrax, 2023-07-19:
[viterbiContent.txt](https://github.com/JVital2013/vitality-goes/files/12099536/viterbiContent.txt)

Hi,

It was too big for PasteBin. I attached it here.

JVital2013, 2023-07-20:
Thanks - I'm looking over all the data, and I was wrong before - data is not making it to graphite (or very little is). Vitality GOES can talk to graphite now, but there's no data home.

The chart you have in your screenshot shows how much data is coming into graphite, and it rounds to 0. Here's what mine looks like:

[image].

**So, the question is why that may be**. My guess is something got corrupted within the graphite database. As root, run the following commands to reset the entire graphite stack:

```bash
docker stop graphite
rm -rf /var/lib/graphite
mkdir -p /var/lib/graphite/config
mkdir -p /var/lib/graphite/data
mkdir -p /var/lib/graphite/statsd_config
mkdir -p /var/lib/graphite/log
docker start graphite
```

Afterwards, wait 10 minutes and see if your charts start showing data

pumatrax, 2023-07-20:
Okay looks like its working. 

After following your steps I started seeing data in graphite but the webpage was broken even worse.  I think I may have recreated the docker without being root last time. I redid that step of removing old docker and readding and it's all working now. 
[image]

Thanks for the help!

closed 2023-07-20 by pumatrax