JVital2013/vitality-goes #45: Question about GOES18 entires

pumatrax, 2023-07-28:
Hi, 

After my latest update I noticed I have some FD entries for GOES18, but they have an error. I dont see a GOES18 folder on my media card so I dont think I am downloading anything for it. 

How can I get this working. I was trying to figure out if I see settings for GOES18 but I not too familiar with the configs to know for sure whats happening. 

[image]

JVital2013, 2023-07-28:
Is your main `config.ini` the same as it was [here](https://github.com/JVital2013/vitality-goes/issues/44#issuecomment-1642557470)?

If so, look below for what the GOES-18 Relay section should look like in `abi.ini` (assuming a semi-standard `goesproc-goes.conf` config. Does yours look similar? The `path` for each is the important bit.

If so, we'll move on to the next troubleshooting step.

```ini
[fdgoes18_16]
path = {GOES16}/goes18/fd/ch13/
title = "GOES 18 - Channel 13 (Clean Longwave IR Relay)"
videoPath = GOES18Ch13.mp4

[fdgoes18e_16]
path = {GOES16}/goes18/fd/ch13_enhanced/
title = "GOES 18 - Channel 13 (Relay; Enhanced)"
videoPath = GOES18Ch13Enhanced.mp4

[fdgoes18Sanchez_16]
path = {GOES16}/goes18/fd/sanchez/
title = "GOES 18 - Sanchez False Color (Relay)"
videoPath = GOES18Sanchez.mp4

[compositeWestEast]
path = {GOES16}/composite/
title = "GOES West/East Sanchez Composite"
videoPath = Composite.mp4
```

pumatrax, 2023-07-29:
Hi, 

Yes, I do have that in my abi.ini.

I am realizing I didnt update my goesproc-goesr.conf file and I am using an older one I am posting below. I tried the last one which I do see handlers for goes18, but I get an error when trying to run goesproc. 

`pi@hampi:~ $ goesproc -c /usr/local/share/goestools/goesproc-goes.conf -m packet --subscribe tcp://127.0.0.1:5004
Invalid image handler origin: goes18
`

`######################
#  General Section   #
######################

# EMWIN
[[handler]]
type = "emwin"
directory = "/media/SDCARD/emwin"

# Images relayed from Himawari-8.
[[handler]]
type = "image"
origin = "himawari8"
directory = "/media/SDCARD/himawari8/{region:short|lower}"
filename = "Himawari8_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
format = "jpg"
json = false

  [[handler.map]]
  path = "/usr/local/share/goestools/ne/ne_50m_admin_0_countries_lakes.json"

  [[handler.map]]
  path = "/usr/local/share/goestools/ne/ne_50m_admin_1_states_provinces_lakes.json"

# NWS text (weather reports).
[[handler]]
type = "text"
origin = "nws"
directory = "/media/SDCARD/nws"
filename = "{time:%Y%m%dT%H%M%SZ}_{awips:nnn}{awips:xxx}"
json = false

# NWS images.
[[handler]]
type = "image"
origin = "nws"
directory = "/media/SDCARD/nws"
filename = "{time:%Y%m%dT%H%M%SZ}_{filename}"
format = "png"
json = false

# Miscellaneous text.
[[handler]]
type = "text"
origin = "other"
directory = "/media/SDCARD/text"
filename = "{time:%Y%m%dT%H%M%SZ}_{filename}"
json = false

####################
#  GOES-15 Relay   #
####################

# Store relayed GOES-15 full disks
[[handler]]
type = "image"
origin = "goes15"
regions = [ "fd" ]
directory = "/media/SDCARD/goes15/{region:short|lower}"
filename = "GOES15_{region:short}_{channel:short}_{time:%Y%m%dT%H%M%SZ}"
[... 737 lines cut ...]
    { units = 193, color = "#ffe300" },
    { units = 213, color = "#81ff00" },
    { units = 233, color = "#00ff23" },
    { units = 253, color = "#00ffbb" },
    { units = 273, color = "#00abff" },
    { units = 293, color = "#0011ff" },
    { units = 313, color = "#7500eb" },
    { units = 321, color = "#000000" },
  ]

  # Cloud Top Height Algorithm (Meters)
    [handler.gradient.ACHA]
   points = [
    { units = 0, color = "#000000" },
    { units = 1500, color = "#5e10b1" },
    { units = 3000, color = "#360ae5" },
    { units = 6050, color = "#0d85e4" },
    { units = 9100, color = "#16f2b4" },
    { units = 12200, color = "#19f108" },
    { units = 15250, color = "#e4ff2b" },
    { units = 19845, color = "#ef630e" },
    { units = 19923, color = "#000000" },
  ]
  
`

JVital2013, 2023-07-29:
If you're getting errors about goes18 being an invalid origin, you must be running an old version of goestools. Goestools was only updated on Mar 16, 2022 to support GOES-18 and GOES-19.

Update goestools by recompiling from https://github.com/pietern/goestools/ and the issue should go away

pumatrax, 2023-07-30:
I update goestool and that did the job. I havent done any updates for over a year. 

Things are working now, thanks again!

I think the final thing is I am now trying to run the Sanchez.sh and I get an error saying: Invalid underlay path:  /tmp/underlay.jpg 

I don't know if that's normal.

Thanks for all your help so far. 

```
[2023-07-30 13:14:37] Creating GOES16_FD_sanchez_20230730T080020Z.jpg...
/home/pi/vitality-goes/scripts-linux/Sanchez.sh: line 87:  9063 Killed                  xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 21600x10800 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
```

JVital2013, 2023-07-30:
Looks like the key line is the one above it, that says `line 87:  9063 Killed xplanet ...`

**The Linux Kernel is killing xplanet for some reason - probably due to not enough RAM on your system**. The Sanchez script works with very high resolution underlays. You can try reducing the underlay resolution from 21600x10800 to 10800x5400 by updating xplanet's `-geometry` argument on:

- [Line 105](https://github.com/JVital2013/vitality-goes/blob/e2c54603edf1fee42af1b60f12722132ecb4fdf7/scripts-linux/Sanchez.sh#L105C148-L105C159)
- [Line 130](https://github.com/JVital2013/vitality-goes/blob/e2c54603edf1fee42af1b60f12722132ecb4fdf7/scripts-linux/Sanchez.sh#L130C148-L130C159)
- [Line 172](https://github.com/JVital2013/vitality-goes/blob/e2c54603edf1fee42af1b60f12722132ecb4fdf7/scripts-linux/Sanchez.sh#L172C150-L172C161)

**Also, it looks like you may have an old version of Sanchez.sh** since xplanet does not get called on line 87 anymore. I'd recommend updating Sanchez.sh

pumatrax, 2023-07-30:
I am running a RPI 4 with 4GB memory. These are some high resolutions so I get it. The computer crashes at times when Sanchez.sh, it might be overheating since it sits out. I did try lowering the resolution but the same is happening. 

I compared my Sanchez.sh with the one here and it's the same. This is line 87: `for srcImg in $(find "$sanchezSrcPath16" -type f)`

My paths looks correct in scriptconfig.ini and is located in the scripts folder where I execute Sanchez.sh. I also noticed that I don't have the /tmp/underlay.jpg in the /tmp folder 

It seems like the xplanet command is a new command after the killed. It sits there for a few seconds then throws the Invalid underlay path: /tmp/underlay.jpg. Then the another jpg is processed and I get the ./Sanchez.sh: line 87:  6466 Killed\

Seems like its two errors. 

```
[2023-07-30 18:13:08] Creating GOES16_FD_sanchez_20230730T150020Z.jpg...
./Sanchez.sh: line 87:  6466 Killed                  xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 10800x5400 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
[2023-07-30 18:13:39] Creating GOES16_FD_sanchez_20230730T153020Z.jpg...
./Sanchez.sh: line 87:  6752 Killed                  xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 10800x5400 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
[2023-07-30 18:14:21] Creating GOES16_FD_sanchez_20230720T080021Z.jpg...
./Sanchez.sh: line 87:  6964 Killed                  xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 10800x5400 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
[2023-07-30 18:14:54] Creating GOES16_FD_sanchez_20230720T083021Z.jpg...
./Sanchez.sh: line 87:  7167 Killed                  xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 10800x5400 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
[2023-07-30 18:15:28] Creating GOES16_FD_sanchez_20230720T090021Z.jpg...
./Sanchez.sh: line 87:  7459 Killed                  xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 10800x5400 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
[2023-07-30 18:15:59] Creating GOES16_FD_sanchez_20230720T093021Z.jpg...
./Sanchez.sh: line 87:  7659 Killed                  xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 10800x5400 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
[2023-07-30 18:16:37] Creating GOES16_FD_sanchez_20230720T100021Z.jpg...
```

pumatrax, 2023-07-31:
Hi, 

I noticed that I was was executing the programs from a different location where this latest goestools build put the files. Before it was /usr/bin, not its /usr/local/bin.  I am not sure if that's related to my problem but I will allow it to download new products first before checking the Sanchez script again. I deleted the old versions. 

Curious, this is how I am executing both goestools executables. I have desktop shortcuts that run both programs from the autostart folder.

```
"/usr/local/bin/goesproc" -c "/usr/local/share/goestools/goesproc-goes.conf" -m packet --subscribe tcp://127.0.0.1:5004
```

```
"/usr/local/bin/goesrecv" -v -i 1 -c "/home/pi/goesrecv.conf"

```

Let me know if that looks good.

pumatrax, 2023-07-31:
Hi,

Yes, it's definitely a memory issue. It's odd because I believe I have created a Sanchez file right after I updated. otherwise, I never used the feature until recently.  I tried to resize those files with convert and the sizes are too big for the imagemagik program to resize.

So, I had a thought....

Increase the swap file. Its now at 1GB and I see the files processing. I even see the underlay.jpg in /tmp folder. 

I guess this should do it on the ticket. Thanks for all your help. 

[image]

JVital2013, 2023-07-31:
That'll work - might be a bit slow, but it'll work!