JVital2013/vitality-goes #47: Update files to work with XRIT Demodulator - Working code examples below

abomb60, 2023-09-24:
### Discussed in https://github.com/JVital2013/vitality-goes/discussions/46

<div type='discussions-op-text'>

<sup>Originally posted by **abomb60** September 23, 2023</sup>
Vitality-goes is really a great frontend for this data but I had a need to make this work with the paid-for XRIT Demodulator on Windows as SatDump hasn't been reliable at all for me and crashes often for no reason. Figured I'd share in case anyone else has the need.

Files to be updated are config.ini, abi.ini, meso.ini and dataHandler.php.  All other weather data seems to pull right in if you have your paths set correctly (admin, emwin, etc).

**Files to update** (paths are unique to my setup so be sure to update for yours)

**config.ini**
[general]
adminPath = "E:\GOES\xrit_summer_22_G18_interleave\products\Text"
emwinPath = "E:\GOES\xrit_summer_22_G18_interleave\products\EMWIN"
[paths]
GOES16 = "E:\GOES\xrit_summer_22_G18_interleave\products"

**abi.ini** (full file below but just need to update path & filter)
[_category_]
title = "Full Disk"
icon = globe-americas

[fdfc_16]
path = "{GOES16}\Full Disk\"
title = "GOES 16 - Color"
filter = "-FSCLR-" 
fast = true

[fdch02_16]
path = "{GOES16}\Full Disk"
title = "GOES 16 - Channel 2 (Red)"
filter = "-VIS"
fast = true

[fdch07_16]
path = "{GOES16}\Full Disk"
title = "GOES 16 - Channel 7 (Shortwave IR)"
filter = "3.9"
fast = true

[fdch08_16]
path = "{GOES16}\Full Disk"
title = "GOES 16 - Channel 8 (Upper Troposphere)"
filter = "6.2"
fast = true

[fdch09_16]
path = "{GOES16}\Full Disk"
title = "GOES 16 - Channel 9 (Mid Troposphere)"
filter = "6.9"
fast = true

[fdch13_16]
path = "{GOES16}\Full Disk"
title = "GOES 16 - Channel 13 (Clean Longwave IR)"
filter = "10.3"
fast = true

[fdch14_16]
path = "{GOES16}\Full Disk"
title = "GOES 16 - Channel 14 (Longwave IR)"
filter = "11.2"
fast = true

[fdch15_16]
path = "{GOES16}\Full Disk"
title = "GOES 16 - Channel 15 (Dirty Longwave IR)"
filter = "12.3"
fast = true

[fd_18]
path = "{GOES16}\Full Disk"
title = "GOES 18 - Channel 13 (Relay)"
filter = "G18-"
fast = true

**meso.ini** (again full file below but just need to update the path and filter for each)
[_category_]
title = "Mesoscale Imagery"
icon = search-plus

[m1fc_16]
path = "{GOES16}\Meso1"
title = "GOES 16 - M1 - False Color"
fast = true
filter = "-FSCLR-"

[m1ch02_16]
path = "{GOES16}\Meso1"
title = "GOES 16 - M1 - Channel 2 (Red)"
fast = true
filter = "-VIS"

[m1ch07_16]
path = "{GOES16}\Meso1"
title = "GOES 16 - M1 - Channel 7 (Shortwave IR)"
fast = true
filter = "3.9"

[m1ch13_16]
path = "{GOES16}\Meso1"
title = "GOES 16 - M1 - Channel 13 (Clean Longwave IR)"
fast = true
filter = "10.3"

[m2fc_16]
path = "{GOES16}\Meso2"
title = "GOES 16 - M2 - False Color"
fast = true
filter = "-FSCLR-"

[m2ch02_16]
path = "{GOES16}\Meso2"
title = "GOES 16 - M2 - Channel 2 (Red)"
fast = true
filter = "-VIS"

[m2ch07_16]
path = "{GOES16}\Meso2"
title = "GOES 16 - M2 - Channel 7 (Shortwave IR)"
fast = true
filter = "3.9"

[m2ch13_16]
path = "{GOES16}\Meso2"
title = "GOES 16 - M2 - Channel 13 (Clean Longwave IR)"
fast = true
filter = "10.3"

**dataHandler.php**
Find the section around line 605 that says "//Check if it's an image metadata request" and put these 2 lines in the appropriate spot.  I modifed the beginz & endz sections.

case "beginz":
				$regex = "/(\\\\|\/)(?<date>[0-9]{12})[^\\\\\/]*{$config['categories'][$_GET['id']]['data'][$_GET['subid']]['filter']}[^\\\\\/]*\..{3}$/i";
				$dateFormat = "YmdHi";
				break;

case "endz":
				$regex = "/{$config['categories'][$_GET['id']]['data'][$_GET['subid']]['filter']}[^\\\\\/]*(?<date>[0-9]{12})\..{3}$/i";
				$dateFormat = "YmdHi";
				break;

Section starting around line 680:
case "beginz":
			$regex = "/(\\\\|\/)" . $DateTime->format('YmdHi') . "[^\\\\\/]*{$config['categories'][$_GET['id']]['data'][$_GET['subid']]['filter']}[^\\\\\/]*\..{3}$/i";
			break;
case "endz":
			$regex = "/{$config['categories'][$_GET['id']]['data'][$_GET['subid']]['filter']}[^\\\\\/]*" . $DateTime->format('YmdHi') . "\..{3}$/i";
			break;

</div>

JVital2013, 2023-09-24:
Thanks! I'll work this into an update. I will change it a bit though so it's compatible across the board. My implementation will probably add new regex within "beginxrit/endxrit" sections to keep from breaking goestools compatibility.

Once I push the changes I'll ask you to test, since I don't have the software. Thanks again!

JVital2013, 2023-09-24:
@abomb60 I already pushed the changes - no release yet, but can you test against the latest main branch? Make sure to update your dataHandler.php, and use the new sample configs in the `/configs` dir to make sure they work.

You'll probably notice the only real difference between my configs and yours is that my changes require adding `mode = "xrit"` to each handler, since my new dataHandler is looking for it.

abomb60, 2023-09-24:
Thanks for looking at this request so quickly!  I will grab the code and test it out later today/tomorrow.

JVital2013, 2023-10-25:
@abomb60 have you had a chance to test the changes?

JVital2013, 2023-12-10:
I'm going to close this for now since my implementation should work, but if there are issues, feel free to let me know!

closed 2023-12-10 by JVital2013