JVital2013/vitality-goes #51: Local Weather Info

ve3elb, 2023-11-01:
Trying to setup the local weather page for buffalo new york but when I put the info I got from the websites I get errors anyone can help with the correct info. thank you

[image]
[image]

ve3elb, 2023-11-01:
@JVital2013  Thank You very much. Am in Toronto Canada so buffalo is the closest to my. however I still want to add that generic Canadian weather info you told me about but I don't remember where and how it has to be done.. Thank You for all your help

JVital2013, 2023-11-01:
On the "Other EMWIN" screen, it looks like there are a number of products that contain Toronto. I'll put screenshots of their configs here.

[image]
[image]
[image]

ve3elb, 2023-11-02:
@JVital2013  How can I add this to the other emwin.ini file. I tried duplicating one of the settings but no luck.

JVital2013, 2023-11-02:
This will do it:

```ini
[NATEMP]
identifier = "TCPNAMUS"
title = "North American Temps"
format = formatted
truncate = 3

[UVINFO]
identifier = "UVIWAOCN"
title = "UV Information"
format = formatted
truncate = 4

[GATEWAYOBS]
identifier = "OPUCHSSC"
title = "[Test] KCHS/KSAV Gateway Observations"
format = formatted
truncate = 3
```

closed 2023-12-10 by JVital2013