JVital2013/vitality-goes #53: Problems with forwarding and port numbers other than 80

RyokoAsakura, 2023-12-06:
I have a dynamic DNS forward set up to get from my outside domain into my home to hit my vitality-goes RPi. I use different port numbers to forward different servers that I run. My vitality-goes RPi receives outside requests by going to http://sat.DNSSERVICENOTREAL.whatever:16000. This gives me the following:
[image]
(Note: There is nothing in the drop-down for the bad data. It's literally blank.)

However, if I change the forward to come in on port 80, everything works as expected. (No need for picture.)

I've taken everything out of the equation except for vitality-goes seeming to expect port 80 being the failure. Everything works fine on the LAN IP as well.

Any help with this issue would be appreciated. Thanks for your time and awesome product!

JVital2013, 2023-12-10:
This sounds like an apache configuration problem - could you please share how you've reconfigured apache to accept requests on 16000?

RyokoAsakura, 2023-12-17:
My local router takes the incoming requests on 16000 and forwards them to the Pi on the Pi's port 80. I'll throw my apache2.conf below. 
`# This is the main Apache server configuration file.  It contains the
# configuration directives that give the server its instructions.
# See http://httpd.apache.org/docs/2.4/ for detailed information about
# the directives and /usr/share/doc/apache2/README.Debian about Debian specific
# hints.
#
#
# Summary of how the Apache 2 configuration works in Debian:
# The Apache 2 web server configuration in Debian is quite different to
# upstream's suggested way to configure the web server. This is because Debian's
# default Apache2 installation attempts to make adding and removing modules,
# virtual hosts, and extra configuration directives as flexible as possible, in
# order to make automating the changes and administering the server as easy as
# possible.

# It is split into several files forming the configuration hierarchy outlined
# below, all located in the /etc/apache2/ directory:
#
#       /etc/apache2/
#       |-- apache2.conf
#       |       `--  ports.conf
#       |-- mods-enabled
#       |       |-- *.load
#       |       `-- *.conf
#       |-- conf-enabled
#       |       `-- *.conf
#       `-- sites-enabled
#               `-- *.conf
#
#
# * apache2.conf is the main configuration file (this file). It puts the pieces
#   together by including all remaining configuration files when starting up the
#   web server.
#
# * ports.conf is always included from the main configuration file. It is
#   supposed to determine listening ports for incoming connections which can be
#   customized anytime.
#
# * Configuration files in the mods-enabled/, conf-enabled/ and sites-enabled/
#   directories contain particular configuration snippets which manage modules,
#   global configuration fragments, or virtual host configurations,
#   respectively.
#
#   They are activated by symlinking available configuration files from their
#   respective *-available/ counterparts. These should be managed by using our
#   helpers a2enmod/a2dismod, a2ensite/a2dissite and a2enconf/a2disconf. See
#   their respective man pages for detailed information.
#
# * The binary is called apache2. Due to the use of environment variables, in
#   the default configuration, apache2 needs to be started/stopped with
#   /etc/init.d/apache2 or apache2ctl. Calling /usr/bin/apache2 directly will not
#   work with the default configuration.

# Global configuration
#

#
# ServerRoot: The top of the directory tree under which the server's
# configuration, error, and log files are kept.
#
# NOTE!  If you intend to place this on an NFS (or otherwise network)
# mounted filesystem then please read the Mutex documentation (available
# at <URL:http://httpd.apache.org/docs/2.4/mod/core.html#mutex>);
# you will save yourself a lot of trouble.
#
# Do NOT add a slash at the end of the directory path.
#
#ServerRoot "/etc/apache2"

[... 126 lines cut ...]
#
# These deviate from the Common Log Format definitions in that they use %O
# (the actual bytes sent including headers) instead of %b (the size of the
# requested file), because the latter makes it impossible to detect partial
# requests.
#
# Note that the use of %{X-Forwarded-For}i instead of %h is not recommended.
# Use mod_remoteip instead.
#
LogFormat "%v:%p %h %l %u %t \"%r\" %>s %O \"%{Referer}i\" \"%{User-Agent}i\"" vhost_combined
LogFormat "%h %l %u %t \"%r\" %>s %O \"%{Referer}i\" \"%{User-Agent}i\"" combined
LogFormat "%h %l %u %t \"%r\" %>s %O" common
LogFormat "%{Referer}i -> %U" referer
LogFormat "%{User-agent}i" agent
# Include of directories ignores editors' and dpkg's backup files,
# see README.Debian for details.

# Include generic snippets of statements
IncludeOptional conf-enabled/*.conf

# Include the virtual host configurations:
IncludeOptional sites-enabled/*.conf

# vim: syntax=apache ts=4 sw=4 sts=4 sr noet
ServerName localhost`

JVital2013, 2023-12-18:
That part of the config looks good. Can you try going to http://sat.dnsservicenotreal.whatever:16000/dataHandler.php?type=preload and see if it loads anything? It should be a bunch of JSON data. 

I have a feeling this is an issue with how you're using your router to re-map the port. There's a few ways this can go wrong - for example, it will not work if your router does not support properly hairpin NAT and you're testing within your network. Unfortunately I cannot help you troubleshoot those issues - too many variables with how different routers are configured. You might have better luck reconfiguring Apache itself to use port 16000, and forward it straight through.

RyokoAsakura, 2023-12-18:
http://sat.dnsservicenotreal.whatever:16000/dataHandler.php?type=preload does indeed show me a ton of JSON.

I expect you are probably correct about the router being the issue. No big deal, but wanted to check all troubleshooting off the list, which included raising an issue here.

JVital2013, 2023-12-18:
Oop, that's not good! If you open the webpage in Chrome/Chromum and open developer tools, are there any warnings in the console?

RyokoAsakura, 2023-12-18:
[image]

Used Edge as it's the only Chromium based thing I have nearby. (I am a vehement Firefox user.)

creinemann, 2023-12-25:
Just Hopping in this, I too am having a similar issue, if I change to any port other than 80, I get the 403 error on goes Vitality. I reloaded using the latest version of vitality on Windows 11.  If I use 80 it works fine, but then I can't run some of the other apps I have running on 80.

creinemann, 2023-12-26:
I should add that the same error applies whether on localhost or internet.

JVital2013, 2023-12-28:
Thanks for your patience - this should be fixed as of 4b2c09a.

@creinemann @RyokoAsakura can you test again with the latest commit? If it works for you, I'll push a new release. I want to make sure it works for more people than just me first!

creinemann, 2023-12-28:
Yes, That did it I installed the new script.js and index.php and it work with port forwarding, in my case 8014

RyokoAsakura, 2023-12-29:
I had to redo the whole thing for some reason, but after deleting my original vitality-goes install, backing up my /var/www/html/config directory and replacing it after re-cloning...it works! :) Thank you for the attention, @JVital2013, and happy to see I wasn't crazy since @creinemann jumped in as well.

For now, I'm going to leave this for @JVital2013 to close, but I consider my issue handled. Thanks again!

JVital2013, 2023-12-29:
Sounds good; I just pushed version 1.5.1 to encourage people to update!

closed 2023-12-29 by JVital2013