JVital2013/vitality-goes #54: Hello

ve3elb, 2024-01-18:
For the past few days I have had an issue with Vitality Goes.  All images from all pages are showing except for the EMWIN Graphics page. It keeps showing (server returned error code 500) I have not change any settings or done anything to the setup. 

Does any one have any ideas what I can do to fix this. 

Running Windows 10

Thank You
Vince. 

[image]

[image]

JVital2013, 2024-02-04:
Hi @ve3elb,

Sorry for the delayed response! This can sometimes happen if too many files are saved in the emwin directory - eventually, it becomes too much, since tens of thousands of files are saved in the emwin file every day. Have you configured the `Cleanup-EmwinText.ps1` script to clean stuff up?

This is just a guess based on what I've seen before; you may get some more info if you set debug = true in config.ini file

ve3elb, 2024-02-07:
@JVital2013 

No I have not configured any cleanup file. Where will that file be located. I have looked and did not see it.

JVital2013, 2024-02-07:
Take a look at the [Additional Scripts](https://github.com/JVital2013/vitality-goes/blob/main/docs/scripts.md) documentation. 

You're probably a bit behind in cleanup right now, so it's probably best to do a one-time cleanup of files. The long and short of it is:

1. Make a copy of the `scripts-windows` folder found in this repo somewhere on your computer
2. Make a copy of scriptconfig.ini from the sample config folder, into the folder you made on step 1
3. In scriptconfig.ini, configure abiSrcDir to be the same as the GOES16 path in the main Vitality GOES config, and emwinSrcDir match emwinPath
4. Right-click on Delete-Old-goestools.ps1 and click "Run with Powershell"
5. Give it a while to delete files - if this has never been run, it might take a while

Once done, you should still have data available in Vitality GOES, but older files have been cleaned up. Now, I'd recommend configuring a scheduled task in Windows to run `Cleanup-EmwinText.ps1` daily, and `Delete-Old-goestools.ps1` weekly

ve3elb, 2024-03-24:
@JVital2013 Thank you for the info it's working. However still have an issue. Maybe you can help. After adding the windows schedule tasks. It's does not work. When the scheduled task runs it only open the files on the desktop. Running the files manually works. 

 [image]
[image]
[image]

JVital2013, 2024-05-26:
Sorry for the delayed response - Can you please share how you're scheduling this that it's not working? Either screenshots or exporting the scheduled task should work.