JVital2013/vitality-goes #55: Current Conditions not working

mattfox27, 2024-03-12:
When I leave the system defaults I get the current conditions, temp and all that...When I put in my info for my city I get some but not all current conditions weather info.

I get the Radar, weather summary, and forecast but nothing else, any insight into why not the other info?

mattfox27, 2024-03-12:
What would be the best way to update dataHandler.php?  Can I just do 'git pull' or will that mess up my current config?

JVital2013, 2024-03-12:
That depends on how you installed it, which I could only guess at.

No matter the system, though, you could just delete the current dataHandler.php from your web server folder, and replace it with this one: https://raw.githubusercontent.com/JVital2013/vitality-goes/main/html/dataHandler.php

mattfox27, 2024-03-12:
I just installed it on a raspberry pi, following the directions on the GitHub page.

JVital2013, 2024-03-12:
Got it - in that case, you can do a `git pull` to update your copy of the git repo, then run `rsync -av --exclude 'config' --exclude 'videos' html/ /var/www/html/ --delete`

closed 2024-05-26 by JVital2013