JVital2013/vitality-goes #58: Is there a way to make a webpage auto display they latest image from a satelite?

mattfox27, 2024-04-20:
I want to setup a kiosk browser on a tablet mounted to my wall to display the latest image from goes 18, is there a way to do this with Vitality?  Or can I put in a feature request?  It would be awesome to have a webpage to goto that would just show the image with black background and it auto updates like every 30min.

Thanks

ImDroided, 2024-05-01:
I am looking for something similar but I have a website already running that I want to be able to bring the latest image into. I run my site on a server here so a script on my webserver would grab latest image from my GOES and NOAA APT servers and a custom page to display them would have to be built. If this was a feature I could just iframe it and be done lol

ImDroided, 2024-05-02:
> > I am looking for something similar but I have a website already running that I want to be able to bring the latest image into. I run my site on a server here so a script on my webserver would grab latest image from my GOES and NOAA APT servers and a custom page to display them would have to be built. If this was a feature I could just iframe it and be done lol
> 
> I saw a script on FB that would send the latest image via FTP and then rename it to latestimage.jpg so I was thinking of setting up something like that. Maybe a corn job that would grab the latest image and rename and send via FTP to website.

You by chance have a link to that script?

ImDroided, 2024-05-02:
Also FTP is not secure at all the best way to do this if you are on the same network would be have a script on the NOAA receiver save the latest image also as latest.jpg or whatever and have the webserver pull that image every 15 minutes or so saving it in the web dir to be displayed on the site. If the web server is offsite then sftp should work if you have ssh access

mattfox27, 2024-05-02:
> You by chance have a link to that script?

`How to - FTP of the newest MCIR-Image to Webserver:
sudo apt-get update
sudo apt-get install ftp
------------------------
edit the script receive_noaa.sh which you'll find here: /home/pi/raspberry-noaa-v2/scripts
go to line 421 and insert then the following code:
# FTP upload
ftp -n <<EOF
open Server-Adress
user USERNAME PASSWORD
passive
put "${IMAGE_FILE_BASE}-MCIR.jpg" newnameofnoaaimage.jpg
bye
EOF`

JVital2013, 2024-05-06:
This is a good idea - I've found myself wanting to do this and ended up bashing the API to get the right image. I'll let you folks know once this is implemented!

mattfox27, 2024-05-06:
That would be so awesome, I have several tablets around my house running fully kiosk browser and it would work awesome with this idea

ImDroided, 2024-05-08:
If we have a url for the latest image then you can also set it as your desktop background. Plus if you have a smart tv you can usually set the screen saver image with a url. Then I can set my webserver to grab that image every 15 minutes or so.

JVital2013, 2024-05-26:
This has been implemented on the current main branch.

See the documentation [here](https://github.com/JVital2013/vitality-goes/blob/main/docs/datahander-api.md#data) for the details, but essentially all you need to do is copy the URL of the image you want, and remove the timestamp parameter from the URL. Without the timestamp, it will always return the latest image.

As an example, this URL will always return the latest GOES-16 False Color image (with the stock configs):

```
https://www.example.com/dataHandler.php?type=data&id=abi&subid=fdfc_16
```

closed 2024-05-26 by JVital2013