JVital2013/vitality-goes #59: Is there a way to display Special Weather Statement (SPS) in Current Weather

Stormtracker017, 2024-04-23:
Just wondering if there is a way to display a Special Weather Statement (SPS) in Current Weather.  Currently I have it displaying in Other EMWIN section.

JVital2013, 2024-05-06:
Yes, I'll build this into the alert system. It looks like they are all geofenced and are commonly used, so implementing this should not be a problem. Honestly, I can't believe I missed it.

I'll let you know when it's done!

Stormtracker017, 2024-05-06:
Thanks....that would be awesome!!!

Stormtracker017, 2024-05-25:
> This has been implemented as of the latest commit. To use it, you can just
> pull the latest dataHandler.php file and replace the copy you have on your
> web server.
>
> This change will, of course, be rolled into the next versioned update as
> well
>
> weather.alerts.png (view on web)

Jamie

Thanks for the update.  Will give the updated file a try.

Rich

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

JVital2013, 2024-05-25:
PS: I just found and fixed a serious bug in the logic that computed if you are within the alert boundary or not (and therefore, if it should be shown).

This time it's just a minor change to functions.php.

Stormtracker017, 2024-05-25:
Thanks...will let you know how the update works.

Stormtracker017, 2024-07-04:
First of all Happy Forth!!!  Currently we have a SPS for our area, but still does not display in the current weather section.  I can retrieve it if I load additional data.  I updated my web server with the latest files.

```
Special Weather Statement
National Weather Service Binghamton NY
534 PM EDT Thu Jul 4 2024

NYZ024-055>057-062-PAZ038>040-043-044-047-048-072-042345-
Chemung-Tioga-Broome-Delaware-Sullivan-Bradford-Susquehanna-
Northern Wayne-Wyoming-Lackawanna-Luzerne-Pike-Southern Wayne-
534 PM EDT Thu Jul 4 2024

...SCATTERED SHOWERS AND THUNDERSTORMS ACROSS NORTHEAST PA AND
THE SOUTHERN TIER OF NY SHOULD BE OVER BY 8 PM...

Scattered showers with isolated thunderstorms will move through
northeast PA and the southern tier of NY between 530 pm and 8 pm.
A few thunderstorms could contain gusty winds and brief heavy
downpours. Most of the shower and thunderstorm activity will be
over by 8 pm.

$$
```

Thanks

Stormtracker017, 2024-07-05:
It seems like the local WFO sends out SPS (SWS) two different ways:

One with only the zone number 
the second with zone number, lat and long.

first way:

[image]

Second way:
[image]

Don't know if that is the reason I don't see the SPS.

Thanks

Stormtracker017, 2024-07-11:
Currently we have a SPS issued for our area, but does not display in the current weather section. I can retrieve it if I load additional data. 

[image]