JVital2013/vitality-goes #60: Fails with New Satdump directory structure

burgeda, 2024-04-28:
Recently satdump made a fairly radical change in the directory structure where images are stored and now instead of basically one big directory with all the ABI images in it they have a new layer that seems to be based upon time and a new one is created every hour.

So instead of

{GOES16}/IMAGES/GOES-16/Full Disk  

They are now in

{GOES16}/IMAGES/GOES-16/Full Disk/2024-04-27_14-30-20

And that final dir keeps changing.

If I put the exact 2024-04-27_14-30-20 in the abi.ini  file it'll find the images, but then within an hour ... it's a different directory.

JVital2013, 2024-04-29:
Hello!

Yes, a lot is going on over at SatDump these days. I am working on an update to Vitality GOES that is compatible with the new SatDump changes. I hope to have a beta available soon, and a new release around the same time as SatDump 1.2.0 is released.

Stay tuned!

ImDroided, 2024-05-01:
What is the latest satdump that still works with this?

burgeda, 2024-05-01:
Hi ImDroided

Sorry I can't tell you the exact build where this stopped.  I do know that starting about a week ... maybe nine days ago they started a HUGE change, a great change from what I can see.  Big things coming in the GOES images.  Good things.

I tend to clone from git and rebuild every time I have a hiccup and I know I rebuilt ... on 5 or 6 of the past days - straight from git so I'm not certain which EXACT one started the changes.

The official 1.1.4 build will work just fine with Vitality-Goes and I have every confidence that once Satdump settles down Vitality-Goes will be updated quickly as well.

Sorry I couldn't be more specific.

JVital2013, 2024-05-06:
Support has been added as of 1.5.2!

closed 2024-05-06 by JVital2013

burgeda, 2024-05-06:
Thanks, I'll give it a whirl this even, after work.

---- On Sun, 05 May 2024 22:49:38 -0400 Jamie Vital ***@***.***> wrote ---

Support has been added as of 1.5.2!