JVital2013/vitality-goes #63: Help setting up GOES16 and 18 feeds together.

ImDroided, 2024-05-08:
I have 2 nooelec dish's one pointed at GOES 16 and the other at 18 is there a way to set this up for both at the same time?

creinemann, 2024-05-08:
[image]
Yes, you need to add GOES 18 paths and imagery to your exiting configs.

```
[paths]
GOES16 = E:/Satellite Imagery
GOES16Arch = H:/GOES_16_2020_2021_2022
GOES17 = E:/Satellite Imagery GOESWEST
GOES18 = E:/Satellite Imagery GOESWEST
HIMAWARI = E:/Satellite Imagery GOESWEST
MeteorM2 = E:/MeteorM2
GOES16_36HOUR = E:/Satellite Related/GOES STUFF/GOES36hours
SDO = E:/GOES  DATA CENTER/SDO_Imagery
GEO = E:/GeoMagnetic
```

and then edit your abi.ini (etc.) to also include the GOES 18 data - here is a snippet of my abi.ini
```
[fdch15_16]
path = {GOES16}/goes16/fd/ch15/
title = "GOES 16 - Channel 15 (Dirty Longwave IR)"
;videoPath = GOES16Ch15.mp4

[fdch15e_16]
path = {GOES16}/goes16/fd/ch15_enhanced/
title = "GOES 16 - Channel 15 (Dirty Longwave IR, Enhanced)"
;videoPath = GOES16Ch15Enhanced.mp4

;[fdgoes17Sanchez_16]
;path = {GOES16}/goes17/fd/sanchez/
;title = "GOES 17 - Sanchez False Color (Relay)"
;videoPath = GOES17Sanchez.mp4

;[composite1617]
;path = {GOES16}/composite/
;title = "GOES 16/17 Sanchez Composite"
;videoPath = Composite.mp4

;videoPath = SDO_193.mp4
[3618HourTiled]
path = {GOES16_36HOUR}/
title = "GOES18 - 36 Hour Imagery"
color = #3b6b68
videoPath = 1836hourstiled.mp4

[fdCUSTOMLUT18]
path = {GOES18}/goes18/fd/CUSTOMLUT/
title = "GOES 18 - USRADIOGUY COLOR LUT"
videoPath = GOES1836Hours.mp4
color = #013A20

[fdch02_18]
path = {GOES18}/goes18/fd/ch02/
title = "GOES 18 - Channel 2 (Red)"
;videoPath = GOES18Ch2.mp4
color = #013A20

[fdch07_18]
path = {GOES18}/goes18/fd/ch07/
title = "GOES 18 - Channel 7 (Shortwave IR)"
;videoPath = GOES18Ch7.mp4
color = #013A20
```

creinemann, 2024-05-08:
Also note that my I config.ini has extra settings that you probably do not have, I am using Vitality GOES for other imagery as well
Carl
usradioguy.om

ImDroided, 2024-05-08:
> Also note that my I config.ini has extra settings that you probably do not have, I am using Vitality GOES for other imagery as well Carl usradioguy.om

Lol man everytime I turn around you have the answer. Thanks again for the help here and on my other issue on FB any chance you  have a writeup on your site or would do a writeup on your site for a dual goes system?

JVital2013, 2024-05-26:
@creinemann's answer really is all there is to it. The way I would recommend getting started with a dual setup is:

1. Configure SatDump/goestools to receive from each satellite, each in their own folder
2. Start setting up Vitality GOES by setting up for GOES-16 only, as normal.
3. Once working, add a GOES18 path in your `config.ini` file, in addition to GOES16 which is already there
4. Open the abi, meso, etc example config files for GOES-18, copy their contents, and paste it at the bottom of your already-configured abi, meso etc files for GOES 16. This will result in the abi, etc files having configs for both GOES-18 and GOES-16.

This is the easiest way to do it - there are many other ways, including making distinct menu items for each satellite. This can be done by having two abi files, two meso files, etc and specify them each under categories in config.ini. You can review the documentation [here](https://github.com/JVital2013/vitality-goes/blob/main/docs/config.md#categories) for more info - there's a lot of permutations you could choose to do.

ImDroided, 2024-05-29:
Thanks for the help I ended up adding to the menu in the end. Should we leave this open for others or close it?

JVital2013, 2024-05-30:
We'll close it since the issue is resolved for you. Closed issues still provide a good resource for people to look at for help - I know I do on other projects!

closed 2024-05-30 by JVital2013