JVital2013/vitality-goes #64: Server error after update on full disk images

ImDroided, 2024-05-08:
Updated now I am getting this error on the full disk images

The server returned bad data: Invalid server config: satdump_geois not a valid file parser mode!

running on a raspberry pi

ImDroided, 2024-05-08:
Ok so I ran the update process again after noticing that my web dir's files were 2 days old but I still have this happening the error is gone now but the images are still missing.  Some of the full disk images loaded before and the same ones load still now. 

[image]

JVital2013, 2024-05-08:
Which version of SatDump are you on? You need to be on one of the recent SatDump nightlies (ie, one from the last week or so) for the new color composites to work

ImDroided, 2024-05-09:
Ok thanks for the response I figured that was it.

closed 2024-05-09 by ImDroided

BigBayWx, 2024-05-09:
Jamie et al;

Speaking of server errors;

I’ve been running for a few months now and getting a lot of server errors with allocated memory issues.  My guess as to the cause of this is not being able to run the cleanup scripts.  I’ve never been able to get them to work as they always return a wrong source error even though I’m pretty sure I’ve got it right.  What I have done instead is every day or two to FTP into the box and manually delete all the old Emwin/Nws and Goes 16 images and that seems to work.  Now suddenly for the last several days I’m getting the memory errors and when I check the status using the Raspberry Pi app on the iPad it’s showing only about 20% of the RAM being used and about the same for disk size.

My guess is also that a lot of the file information is cached somewhere as if I pull up something like my local area forecast discussion it shows a lot of them as well as an old warning about a month old still on the local conditions page.  Just where is this information cached in the system?  If I can find those files I can manually delete them even if I can’t get the cleanup scripts to work.

Thanks in advance and once again a great program.

Doug

The server returned bad alert data: 

Fatal error: Allowed memory size of 134217728 bytes exhausted (tried to allocate 4096 bytes) in /var/www/html/functions.php on line 241