JVital2013/vitality-goes #67: Error with Sanchez.sh

mattfox27, 2024-05-15:
Im trying to run Sanchez.sh and I keep getting this error

"Invalid underlay path: /tmp/"

JVital2013, 2024-05-26:
Try manually deleting /tmp/underlay.jpg and try again. If that doesn't do it, can you paste the entire output of the script?

chuckdubdub, 2024-07-22:
I'm also having this same issue. There is no file here: /tmp/underlay.jpg

Example of script output:

```
[2024-07-21 22:55:42] Creating GOES16_FD_sanchez_20240718T093021Z.jpg...
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
./Sanchez.sh: line 87:  8833 Aborted                 xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 21600x10800 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
[2024-07-21 22:56:00] Creating GOES16_FD_sanchez_20240718T073021Z.jpg...
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
./Sanchez.sh: line 87:  8864 Aborted                 xplanet -config "$(dirname "$(readlink -fm "$0")")/Resources/xplanet-$satelliteMonth" -body earth -projection rectangular -num_times 1 -geometry 21600x10800 -date $thisDate.$thisTime -output /tmp/underlay.jpg
Invalid underlay path: /tmp/underlay.jpg
```