JVital2013/vitality-goes #74: only the Local Settings option is showing up

nmoon88, 2024-06-04:
### Discussed in https://github.com/JVital2013/vitality-goes/discussions/71

<div type='discussions-op-text'>

<sup>Originally posted by **nmoon88** June  2, 2024</sup>
At the moment despite supposedly having all the directories configured correctly I am only seeing the Local Settings and System Information options on the web page that the server generates. I followed the instructions in the readme file involving the syntax but am still having problems. The files are being dumped by goestools through goesproc with SatDump being used as the de modulator onto an external hard drive the location of which is known. Am I missing something or is there some sort of weirdness with external hard drives?</div>

ImDroided, 2024-06-05:
Did you edit and copy the proper config file examples for your config folder in your web directory? Sounds like you are missing your config files.

nmoon88, 2024-06-05:
> Did you edit and copy the proper config file examples for your config folder in your web directory? Sounds like you are missing your config files.

The weird thing is that I did. I put all the necessary files for GOES 16 into the /var/www/config folder and edited config.ini to point to the root of the drive that the files are being dumped into along with editing the other files to point to the other folders where the files are being stored.

ImDroided, 2024-06-05:
Possibly a file permissions issue maybe? can you post a screenshot of what you are seeing on the website? From what I know if the .ini files are not found/readable it will not show up on the menu of the website.

nmoon88, 2024-06-05:
> Possibly a file permissions issue maybe? can you post a screenshot of what you are seeing on the website? From what I know if the .ini files are not found/readable it will not show up on the menu of the website.

[image]

That is the page as I am seeing it. The only menu options are Local Settings and System Info.

ImDroided, 2024-06-05:
post your file permissions for the ini files

BigBayWx, 2024-06-05:
I had the exact same thing a few months ago using the latest build of Bookworm.  After trying everything else I decided what the heck and tried flashing with Bullseye and it started to work and populate.  I don’t know why but if you have a spare SD card flash it with Bullseye and do the build again and see what happens.

nmoon88, 2024-06-05:
> post your file permissions for the ini files
-rw-r--r-- 1 root root 629 Jun  4 15:52 config.ini
-rw-r--r-- 1 root root 2758 Jun  4 15:59 abi.ini

Same file permissions and I think I see it, it looks like I need to change the permissions to myself and the user associated with Apache2.

ImDroided, 2024-06-05:
yup you got it do a chown on that including group and you should be set

ImDroided, 2024-06-05:
sudo chown -R noaa:www-data /var/www/vitality-goes

change your user and www dir and this hopefully fixes it

nmoon88, 2024-06-06:
> sudo chown -R noaa:www-data /var/www/vitality-goes
> 
> change your user and www dir and this hopefully fixes it

It did and it did not, the other menu options are appearing but I am getting this error message despite having the correct root directory specified in config.ini "The server returned bad data: Invalid server config: path for this image type does not exist"

ImDroided, 2024-06-06:
give me a output of ls -lia /var/www/ and the same for your goes output dir. Its still a permissions issue if your path is correct. make sure you do not have the trailing / in config.ini

ImDroided, 2024-06-06:
You need to chown the goes output dir also or make sure it is readable by www-data

```
noaa@noaa:/srv $ ls -lia
total 80
    25 drwxr-xr-x  6 root root      4096 May 27 20:13 .
     2 drwxr-xr-x 18 root root      4096 May 27 22:15 ..
169008 drwxrwxr-x  4 noaa noaa      4096 Apr 13 19:05 audio
168996 drwxrwxr-x  3 noaa www-data 57344 Jun  5 19:44 images
177503 drwxr-xr-x  4 noaa www-data  4096 Jun  5 17:19 live_output
169026 drwxrwxr-x  2 noaa www-data  4096 Apr 13 19:05 videos
```

nmoon88, 2024-06-06:
> give me a output of ls -lia /var/www/ and the same for your goes output dir. Its still a permissions issue if your path is correct. make sure you do not have the trailing / in config.ini

21107535 drwxr-xr-x  4 nicholas nicholas 4096 Jun  5 19:39 .
20578305 drwxr-xr-x 15 root     root     4096 Jun  1 13:24 ..
21107777 drwxrwxrwx 13 nicholas nicholas 4096 Jun  4 15:59 config
21103346 drwxrwxr-x 10 nicholas www-data 4096 Jun  5 20:13 html

ImDroided, 2024-06-06:
your config folder should be in your html folder and have the same permissions as html unless that is a different config folder for something else

[image]

I have more ini files so ignore those but does your config dir look like this?

```
noaa@noaa:/srv $ ls -lia /var/www/vitality-goes/config/
total 80
179245 drwxrwxr-x  2 noaa www-data 4096 Jun  4 13:29 .
179229 drwxr-xr-x 10 noaa www-data 4096 Jun  4 14:13 ..
179479 -rw-r--r--  1 noaa www-data 4929 May 28 22:18 abi2.ini
168899 -rw-rw-r--  1 noaa www-data 4011 Jun  3 01:40 abi.ini
168817 -rw-rw-r--  1 noaa www-data  845 Jun  3 23:52 config.ini
155475 -rw-r--r--  1 noaa www-data 4933 May 29 12:57 emwin2.ini
155479 -rw-rw-r--  1 noaa www-data 4961 Jun  2 22:15 emwin.ini
179247 -rw-rw-r--  1 noaa www-data   14 May 27 20:17 .htaccess
168907 -rw-r--r--  1 noaa www-data  391 Jun  3 18:20 latest.ini
179477 -rw-r--r--  1 noaa www-data 2577 May 28 21:56 meso2.ini
179365 -rw-rw-r--  1 noaa www-data 2577 May 28 21:56 meso.ini
168893 -rw-r--r--  1 noaa www-data  199 Jun  3 17:35 meteor.ini
168932 -rw-r--r--  1 noaa www-data  186 Jun  4 13:29 noaa.ini
179369 -rw-rw-r--  1 noaa www-data 2792 May 27 20:18 nws.ini
179370 -rw-rw-r--  1 noaa www-data  518 May 27 20:18 otheremwin.ini
179366 -rw-rw-r--  1 noaa www-data 5085 May 28 23:20 scriptconfig.ini
168906 -rw-r--r--  1 noaa www-data    0 Jun  3 18:17 solar.ini
```
run the following but change the commands to your setup this will fix your output dir, fix vitality, and add your user to the www-data group

sudo chown -R noaa:www-data /srv/live_output

sudo chown -R noaa:www-data /var/www/vitality-goes

sudo usermod -a -G www-data nicholas

sudo systemctl restart apache2

Those 4 commands hopefully takes care of that

nmoon88, 2024-06-06:
I did those four commands replacing /srv/live_output with the root of the external drive my data is going to and replacing /var/www/vitality-goes with /var/www/html and it did not work. I still don't have any images appearing

ImDroided, 2024-06-06:
on the webpage is the emwin images loading or the NWS stuff?

nmoon88, 2024-06-06:
neither EMWIN nor NWS stuff is loading, still exactly the same error.

ImDroided, 2024-06-06:
post your config.ini and abi.ini file contents let me see if I can see anything obvious.

nmoon88, 2024-06-06:
config.ini
[general]
;siteTitle = "My GOES-16 Data"
;siteTheme = "light"
;graphiteAPI    = http://127.0.0.1:8080/render/
;adminPath = /media/nicholas/GOES/text
emwinPath = /media/nicholas/GOES/emwin
fastEmwin = false
spaceWeatherAlerts = false
showSysInfo = true
debug = false

[paths]
GOES16 = /media/nicholas/GOES

[categories]
abi = abi.ini
meso = meso.ini
l2 = l2.ini
nws = nws.ini
emwin = emwin.ini

\[_category_]
title = "Full Disk"
icon = globe-americas

[fdfc_16]
path = {GOES16}/goes16_new/fd/fc/
title = "GOES 16 - Color"
;videoPath = GOES16FalseColor.mp4

;[fdsanchez_16]
;path = {GOES16}/goes16_new/fd/sanchez/
;title = "GOES 16 - Sanchez False Color"
;color = #00391D
;videoPath = GOES16Sanchez.mp4

[fdch02_16]
path = {GOES16}/goes16_new/fd/ch02/
title = "GOES 16 - Channel 2 (Red)"
;videoPath = GOES16Ch2.mp4

[fdch07_16]
path = {GOES16}/goes16_new/fd/ch07/
title = "GOES 16 - Channel 7 (Shortwave IR)"
;videoPath = GOES16Ch7.mp4

[fdch07e_16]
path = {GOES16}/goes16_new/fd/ch07_enhanced/
title = "GOES 16 - Channel 7 (Shortwave IR, Enhanced)"
;videoPath = GOES16Ch7Enhanced.mp4

[fdch08_16]
path = {GOES16}/goes16_new/fd/ch08/
title = "GOES 16 - Channel 8 (Upper Troposphere)"
;videoPath = GOES16Ch8.mp4

[fdch08e_16]
path = {GOES16}/goes16_new/fd/ch08_enhanced/
title = "GOES 16 - Channel 8 (Upper Troposphere, Enhanced)"
;videoPath = GOES16Ch8Enhanced.mp4

[fdch09_16]
path = {GOES16}/GOES/goes16_new/fd/ch09/
title = "GOES 16 - Channel 9 (Mid Troposphere)"
;videoPath = GOES16Ch9.mp4

[fdch09e_16]

ImDroided, 2024-06-06:
Have you updated satdump? here is my config.ini and abi.ini for goes satdump

config.ini
```
[general]
;siteTitle = "My GOES-16 Data"
;siteTheme = "light"
satdumpAPI = http://127.0.0.1:8080/api
adminPath = "/srv/live_output/GOES16/Admin Messages"
emwinPath = "/srv/live_output/GOES16/EMWIN"
fastEmwin = true
spaceWeatherAlerts = false
showSysInfo = true
debug = false

[paths]
GOES16 = "/srv/live_output/GOES16"
GOES18 = "/srv/live_output/GOES18"
NOAA = "/srv/images"
METEOR = "/srv/images"

[categories]
latest = latest.ini
abi = abi.ini
abi2 = abi2.ini
noaa = noaa.ini
meteor = meteor.ini
meso = meso.ini
meso2 = meso2.ini
nws = nws.ini
emwin = emwin.ini
;emwin2 = emwin2.ini
solar = solar.ini

[location]
radarCode = GRTLK
stateAbbr = IL
wxZone = ILZ013
orig = LOTIL
rwrOrig = LOTIL
city = Dupage Airport
lat = 44.03
lon = -88.07
timezone = America/Chicago

[otheremwin]
ini = otheremwin.ini
allowUserLoader = true
maxUserFiles = 1000
```
abi.ini

```
[_category_]
title = "GOES 16 Full Disk"
icon = globe-americas

[fdfc_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Color"
filter = "GEO_False_Color_map"
mode = "satdump_geo"
videoPath = GOES16FalseColor.mp4

[fdfc_16nb]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Color (No borders)"
filter = "GEO_False_Color"
mode = "satdump_geo"

[fdmicro_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Night Microphysics"
filter = "Nighttime_Microphysics_map"
mode = "satdump_geo"

;[fdmerge_16]
;path = "{GOES16}/IMAGES/GOES-16/Full Disk"
;title = "GOES 16 - False Color IR Merge"
;filter = "GEO_False_Color_IR_Merge_map"
;mode = "satdump_geo"

;[fdfog_16]
;path = "{GOES16}/IMAGES/GOES-16/Full Disk"
;title = "GOES 16 - Fog Brightness Temperature Difference"
;filter = "Fog_Brightness_Temperature_Difference_map"
;mode = "satdump_geo"

;[fdconv_16]
;path = "{GOES16}/IMAGES/GOES-16/Full Disk"
;title = "GOES 16 - Day Cloud Convection"
;filter = "Day_Cloud_Convection_map"
;mode = "satdump_geo"

[fdrocket_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Day Rocket Plume"
filter = "Day_Rocket_Plume_Detection_map"
mode = "satdump_geo"

;[fd789_16]
;path = "{GOES16}/IMAGES/GOES-16/Full Disk"
;title = "GOES 16 - 789 RGB"
;filter = "789"
;mode = "satdump_geo"

;[fd71315_16]
;path = "{GOES16}/IMAGES/GOES-16/Full Disk"
;title = "GOES 16 - 7/13/15 RGB"
;filter = "7_13_15"
;mode = "satdump_geo"

[fdch02_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 2 (Red)"
filter = "_2_"
;videoPath = GOES16Ch2.mp4

[fdch07_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 7 (Shortwave IR)"
filter = "_7_"
;videoPath = GOES16Ch7.mp4

[fdch07e_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 7 (Enhanced)"
filter = "Shortwave_Window_Band_map"
mode = "satdump_geo"
;videoPath = GOES16Ch7Enhanced.mp4

[fdch08_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 8 (Upper Troposphere)"
filter = "_8_"
;videoPath = GOES16Ch8.mp4

[fdch08e_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 8 (Enhanced)"
filter = "Upper-Level_Tropospheric_Water_Vapor_map"
mode = "satdump_geo"
;videoPath = GOES16Ch8Enhanced.mp4

[fdch09_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 9 (Mid Troposphere)"
filter = "_9_"
;videoPath = GOES16Ch9.mp4'

[fdch09e_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 9 (Enhanced)"
filter = "Mid-level_Tropospheric_Water_Vapor_map"
mode = "satdump_geo"
;videoPath = GOES16Ch9Enhanced.mp4

[fdch13_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 13 (Clean Longwave IR)"
filter = "_13_"
;videoPath = GOES16Ch13.mp4

[fdch13e_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 13 (Enhanced)"
filter = "Clean_Longwave_IR_Window_Band_map"
mode = "satdump_geo"
;videoPath = GOES16Ch13Enhanced.mp4

[fdch14_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 14 (Longwave IR)"
filter = "_14_"
;videoPath = GOES16Ch14.mp4

[fdch14e_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 14 (Enhanced)"
filter = "Infrared_Longwave_Window_Band_map"
mode = "satdump_geo"
;videoPath = GOES16Ch14Enhanced.mp4

[fdch15_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 15 (Dirty Longwave IR)"
filter = "_15_"
;videoPath = GOES16Ch15.mp4

[fdch15e_16]
path = "{GOES16}/IMAGES/GOES-16/Full Disk"
title = "GOES 16 - Channel 15 (Enhanced)"
filter = "Dirty_Longwave_Window_map"
mode = "satdump_geo"
;videoPath = GOES16Ch15Enhanced.mp4

[fd_18]
path = "{GOES16}/IMAGES/GOES-18/Full Disk"
title = "GOES 18 - Channel 13 (Relay)"
filter = "_13_"
;videoPath = GOES18Ch13.mp4

[fde_18]
path = "{GOES16}/IMAGES/GOES-18/Full Disk"
title = "GOES 18 - Channel 13 (Relay, Enhanced)"
filter = "Clean_Longwave_IR_Window_Band_map"
mode = "satdump_geo"
;videoPath = GOES18Ch13Enhanced.mp4
```

nmoon88, 2024-06-06:
I am using goesproc to actually write the data and SatDump is only being used as a demodulator to convert the incoming signal into packets.

ImDroided, 2024-06-06:
Im not sure then what the issue is. it seems like a folder/permission issue. Goes reception in satdump has had a major rework in the last month or so if your fully updated and using a rtl-sdr you can try my config and abi then run 

`satdump live goes_hrit /srv/live_output/GOES16 --source rtlsdr --source_id 00000001 --samplerate 2.4e6 --frequency 1694.1e6 --gain 44.5 --bias --http_server 0.0.0.0:8080`

let that run for an hour or so and see if anything changes.

nmoon88, 2024-06-06:
I did that and nothing changed. I tried goestools plus SatDump and the internal demodulator in SatDump along with specifying the folders where the files are placed (incidentally my fingers hurt for a while from all the typing) and I am still getting the same error about files not being found.

ImDroided, 2024-06-07:
If you did the satdump method you would have needed those config files in the sample config folder. you did this right?  My only suggestion to you would be to start over fresh and follow my guide at https://radiomenace.com/installing-vitalitygoes-on-raspinoaa-image/ and see where that gets you.  Don't worry if you are not using that image the set of commands and configs are what's important.. its either a config issue or permissions issue and I am pretty sure we fixed the permissions issue.

JVital2013, 2024-06-07:
This is definitely a permissions issue in reading the saved imagery/data. It looks like you're saving data to `/media/nicholas/GOES`. In my experience, these auto-mounted drives can ONLY be written to by the user it's mounted by. In this case, it's nicholas, which means www-data cannot access it. Changing permissions here is usually futile as they get re-set by the service that mounts the drive.

Your best bet is to mount the drive via fstab and set permissions correctly. For a permanent setup, this is a better idea anyway. Auto-mount locations can sometimes wander, and before you know it you're not writing data to your external drive anymore, but a folder on your local disk.

nmoon88, 2024-06-08:
That explains it, the weirdness with external drives. I have some spare 1 TB SATA drives kicking around and I think I could physically put one of them in as an internal drive as of course 2.5 inch and 3.5 inch SATA drives use exactly the same connectors and my spare drives are 2.5 inch. I should have known because I encountered this very thing before where an external drive wandered around and I was writing data in weird spots.

JVital2013, 2024-06-12:
I'm glad to hear it's working! The cleanup scripts contained in the repo should help you keep the drive utilization down.

closed 2024-06-12 by JVital2013