JVital2013/vitality-goes #78: Display multiple hazards in current weather

Stormtracker017, 2024-06-23:
As of right now, our area is under a Heat Advisory and a Severe Thunderstorm Watch.  The Heat Advisory was present in the Current Weather, but once the Severe Thunderstorm Watch was issued the Advisory disappeared. Is there away to display multiple hazards at once and then remove them when they expire.

JVital2013, 2024-06-23:
What you are describing is how it's supposed to work. If active alerts are disappearing, you've found a bug.

Try turning on debug in config.ini to see if it spits out any messages (if the alerts are still active, of course)!

Stormtracker017, 2024-06-24:
Of course the Heat Advisory just expired @ 8:00PM, so there is only one.  Next time I have more then one hazards, I'll turn on the debug and see if any messages spit out.

Stormtracker017, 2024-06-24:
Don't know if this is a bug or OHS, but the Severe Thunderstorm Watch which expired at 10:00 PM tonight is still being displayed.  I did clear the cache for the browser, thinking it was a caching issue.  I set the debug to true in the config.ini file, but where do I find the actual log  file?

JVital2013, 2024-06-24:
There is no log - debug pretty much just turns on errors, which display with the alerts.

If nothing is coming up, I will need your exact location config to debug (including lat/lon). Feel free to email me if you wish to keep the information private!

Stormtracker017, 2024-06-24:
Jamie

Thanks for looking into this.  Here is my location info:

[location]
radarCode = NTHES
stateAbbr = PA
wxZone = PAZ047
orig = BGMNY
rwrOrig = BGMNY
city = HAZLETON
lat = 41.07
lon = -75.95
timezone = America/New_York

If you need any additional information, just let me know.

Thanks
Rich

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Stormtracker017, 2024-06-24:
Sent the information you requested.

Stormtracker017, 2024-06-26:
Currently both the Thunderstorm Watch and Warning for the area are being displayed as expected.  Will still monitor to see if the issue rises again.

[image]

Stormtracker017, 2024-07-09:
As of right now, our area is under a Heat Advisory until 8:00 PM and an additional Heat Advisory was just issued for tomorrow. 

[image]

The Current Weather Tab did display the first Heat Advisory but only displays the second Heat Advisory even though the Heat Advisory is still valid until 8:00 PM.

[image]

Stormtracker017, 2024-07-10:
As of right now, our area is under a Heat Advisory until 8:00 PM and an Tornado Watch until 9:00 PM. 

[image]

The Current Weather Tab did display the Heat Advisory but now only displays the Tornado Watch. 

[image]

I do have the debug set to true and don't see any errors.

Stormtracker017, 2024-07-15:
Couple more examples:

[image]

[image]