JVital2013/vitality-goes #88: Current Weather, Other EMWIN, Hurricane Center, and Local Settings not showing up

robert5603, 2024-08-11:
The server can display images in the full disk, non-CMIP, mesoscale, EMWIN, and NWS graphics sections. I also notice the system info section working just fine as well. However, I receive this error message when attempting to load the current weather section displaying alerts:
[image]

**Error message for Current Weather:**

<br />
<b>Fatal error</b>:  Uncaught UnexpectedValueException: RecursiveDirectoryIterator::__construct(/home/pi/.pki): failed to open dir: Permission denied in /var/www/html/functions.php:241
Stack trace:
#0 [internal function]: RecursiveDirectoryIterator-&gt;__construct('/home/pi/.pki', 4128)
#1 /var/www/html/functions.php(241): RecursiveDirectoryIterator-&gt;getChildren()
#2 /var/www/html/dataHandler.php(1341): scandir_recursive('/home/pi', false)
#3 {main}
  thrown in <b>/var/www/html/functions.php</b> on line <b>241</b><br />

**Error message in Other EMWIN:**

<br />
<b>Fatal error</b>:  Uncaught UnexpectedValueException: RecursiveDirectoryIterator::__construct(/home/pi/.pki): failed to open dir: Permission denied in /var/www/html/functions.php:241
Stack trace:
#0 [internal function]: RecursiveDirectoryIterator-&gt;__construct('/home/pi/.pki', 4128)
#1 /var/www/html/functions.php(241): RecursiveDirectoryIterator-&gt;getChildren()
#2 /var/www/html/dataHandler.php(287): scandir_recursive('/home/pi', false)
#3 {main}
  thrown in <b>/var/www/html/functions.php</b> on line <b>241</b><br />

**Error message in Hurricane Center:**

The server returned bad data:<br />
<b>Fatal error</b>:  Uncaught UnexpectedValueException: RecursiveDirectoryIterator::__construct(/home/pi/.pki): failed to open dir: Permission denied in /var/www/html/functions.php:241
Stack trace:
#0 [internal function]: RecursiveDirectoryIterator-&gt;__construct('/home/pi/.pki', 4128)
#1 /var/www/html/functions.php(241): RecursiveDirectoryIterator-&gt;getChildren()
#2 /var/www/html/dataHandler.php(1097): scandir_recursive('/home/pi', false)
#3 {main}
  thrown in <b>/var/www/html/functions.php</b> on line <b>241</b><br />

**Error message in Local Settings:**

The server returned bad data: <br />
<b>Fatal error</b>:  Uncaught UnexpectedValueException: RecursiveDirectoryIterator::__construct(/home/pi/.pki): failed to open dir: Permission denied in /var/www/html/functions.php:241
Stack trace:
#0 [internal function]: RecursiveDirectoryIterator-&gt;__construct('/home/pi/.pki', 4128)
#1 /var/www/html/functions.php(241): RecursiveDirectoryIterator-&gt;getChildren()
#2 /var/www/html/dataHandler.php(750): scandir_recursive('/home/pi', false)
#3 {main}
  thrown in <b>/var/www/html/functions.php</b> on line <b>241</b><br />

If there are additional files/resources you need, I am more than happy to send them your way.

JVital2013, 2024-09-23:
This looks like a configuration error - it is trying to scan your entire /home/pi directory for EMWIN files, which is not what you want. Make sure `emwinPath` in your config.ini is set to the emwin folder directly.

JVital2013, 2025-04-09:
I'm going to close this for now, but let me know if you have more issues!

closed 2025-04-09 by JVital2013