JVital2013/vitality-goes #91: Sanchez.sh / Multiple Tasks / Cache Growing

parsoli, 2024-11-26:
Hey All,

I have Vitality running on a 1socket/4cpu/8gb memory Ubuntu VM.  All files from GoesProc write to a highspeed NAS and Vitility also mounts on same file system on NAS for processing images.

Every few days the CPU starts running over 95% utilized on Vitality, the sanchez image cache grows and I get multiple tasks of sanchez.sh running (which I do understand is normal).

What would cause the cache to grow?

nbkhwjm, 2024-11-27:
Do you have any crontabs?

parsoli, 2024-11-27:
i do:

SHELL=/bin/sh
PATH=/usr/local/sbin:/usr/local/bin:/sbin:/bin:/usr/sbin:/usr/bin

# m h dom mon dow user  command
17 *    * * *   root    cd / && run-parts --report /etc/cron.hourly
25 6    * * *   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.daily )
47 6    * * 7   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.weekly )
52 6    1 * *   root    test -x /usr/sbin/anacron || ( cd / && run-parts --report /etc/cron.monthly )
#

parsoli, 2024-11-27:
[image]

parsoli, 2024-12-02:
> Do you have any crontabs?

So this is the crontab that runs per the instructions posted in GitHub.  I tried to follow the example word for word.  Dumping output to log files.  Anything wrong with this configuration?

PATH=/usr/local/bin:/usr/bin:/bin
SHELL=/bin/bash
0 0 * * *  /home/sparsons/vitality-goes/scripts-linux/CreateVideos-ABI.sh > /home/sparsons/croncreatevideosabi.log 2>&1
0 2 * * *  /home/sparsons/vitality-goes/scripts-linux/CreateVideos-EMWIN.sh > /home/sparsons/croncreatevideosemwin.log 2>&1
55 11 * * *  /home/sparsons/vitality-goes/scripts-linux/Cleanup-EmwinText.sh > /home/sparsons/cronemwincleanup.log 2>&1
25,55 2-23 * * *  /home/sparsons/vitality-goes/scripts-linux/Sanchez.sh > /home/sparsons/cronsanchez.log 2>&1

parsoli, 2024-12-11:
I resolved my issue by doing the following:

Removed Sanchez.sh from the crontab as the crontab was spawning multiple processes based on the 25,55 2-23 timing
Created a Linux service called sanchez.service that runs the Sanchez.sh that runs at startup.  Considering another [service] option that always keeps it running/restarts it in case it bombs out.  One reason I can think of this is in case the NAS that holds the images from GoesProc reboots and severs the SMB mnt connection from VitalityGoes from the NAS

[Unit]
Description=Starts the GoesVitality Sanchez.sh file
After=network.target

[Service]
WorkingDirectory=/home/xxxxxx/vitality-goes/scripts-linux
ExecStartPre=/bin/sleep 30
ExecStart=/home/xxxxxx/vitality-goes/scripts-linux/Sanchez.sh
RemainAfterExit=true

[Install]

WantedBy=multi-user.target

closed 2024-12-11 by parsoli