JVital2013/vitality-goes #92: Current Weather: The Server Returned Bad Data

RyokoAsakura, 2025-03-14:
Unsure what might be causing this, whether it's the NWS weather summary file being bad, or whether it's a local config issue (I don't think it is...)

[image]

Here's the whole output you see in the image:
`
<br />
<b>Notice</b>:  Undefined offset: 16 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 17 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 18 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 19 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 20 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 21 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 22 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 23 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 24 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 25 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 26 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 27 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 28 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 29 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 30 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 31 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 32 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 33 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 34 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 35 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 36 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
<br />
<b>Notice</b>:  Undefined offset: 37 in <b>/var/www/html/dataHandler.php</b> on line <b>1586</b><br />
{"city":"SELINSGROVE","state":"PA","localRadarMetadata":{"images":[{"description":"Rendered: March 11, 2025 12:30 AM EDT","timestamp":1741667436},{"description":"Rendered: March 11, 2025 12:30 AM EDT","timestamp":1741667436},{"description":"Rendered: March 11, 2025 12:45 AM EDT","timestamp":1741668336},{"description":"Rendered: March 11, 2025 1:00 AM EDT","timestamp":1741669235},{"description":"Rendered: March 11, 2025 1:15 AM EDT","timestamp":1741670133},{"description":"Rendered: March 11, 2025 1:30 AM EDT","timestamp":1741671034},{"description":"Rendered: March 11, 2025 1:45 AM EDT","timestamp":1741671938},{"description":"Rendered: March 11, 2025 2:00 AM EDT","timestamp":1741672834},{"description":"Rendered: March 11, 2025 2:15 AM EDT","timestamp":1741673731},{"description":"Rendered: March 11, 2025 2:30 AM EDT","timestamp":1741674631},{"description":"Rendered: March 11, 2025 2:30 AM EDT","timestamp":1741674632},{"description":"Rendered: March 11, 2025 2:45 AM EDT","timestamp":1741675533},{"description":"Rendered: March 11, 2025 2:45 AM EDT","timestamp":1741675536},{"description":"Rendered: March 11, 2025 3:00 AM EDT","timestamp":1741676433},{"description":"Rendered: March 11, 2025 3:15 AM EDT","timestamp":1741677331},{"description":"Rendered: March 11, 2025 3:30 AM EDT","timestamp":1741678234},{"description":"Rendered: March 11, 2025 3:45 AM EDT","timestamp":1741679131},{"description":"Rendered: March 11, 2025 5:15 AM EDT","timestamp":1741684530},{"description":"Rendered: March 11, 2025 5:30 AM EDT","timestamp":1741685431},{"description":"Rendered: March 11, 2025 5:45 AM EDT","timestamp":1741686335},{"description":"Rendered: March 11, 2025 6:00 AM EDT","timestamp":1741687237},{"description":"Rendered: March 11, 2025 6:15 AM EDT","timestamp":1741688133},{"description":"Rendered: March 11, 2025 6:30 AM EDT","timestamp":1741689034},{"description":"Rendered: March 11, 2025 6:45 AM EDT","timestamp":1741689932},{"description":"Rendered: March 11, 2025 7:00 AM EDT","timestamp":1741690840},{"description":"Rendered: March 11, 2025 7:15 AM EDT","timestamp":1741691731},{"description":"Rendered: March 11, 2025 7:30 AM EDT","timestamp":1741692634},{"description":"Rendered: March 11, 2025 7:45 AM EDT","timestamp":1741693530},{"description":"Rendered: March 11, 2025 8:00 AM EDT","timestamp":1741694432},{"description":"Rendered: March 11, 2025 8:15 AM EDT","timestamp":1741695332},{"description":"Rendered: March 11, 2025 8:30 AM EDT","timestamp":1741696230},{"description":"Rendered: March 11, 2025 8:30 AM EDT","timestamp":1741696230},{"description":"Rendered: March 11, 2025 8:45 AM EDT","timestamp":1741697134},{"description":"Rendered: March 11, 2025 9:00 AM EDT","timestamp":1741698033},{"description":"Rendered: March 11, 2025 9:15 AM EDT","timestamp":1741698930},{"description":"Rendered: March 11, 2025 9:30 AM EDT","timestamp":1741699834},{"description":"Rendered: March 11, 2025 9:45 AM EDT","timestamp":1741700733},{"description":"Rendered: March 11, 2025 10:00 AM EDT","timestamp":1741701630},{"description":"Rendered: March 11, 2025 10:00 AM EDT","timestamp":1741701630},{"description":"Rendered: March 11, 2025 10:15 AM EDT","timestamp":1741702530},{"description":"Rendered: March 11, 2025 10:30 AM EDT","timestamp":1741703433},{"description":"Rendered: March 11, 2025 10:45 AM EDT","timestamp":1741704331},{"description":"Rendered: March 11, 2025 11:00 AM EDT","timestamp":1741705234},{"description":"Rendered: March 11, 2025 11:15 AM EDT","timestamp":1741706131},{"description":"Rendered: March 11, 2025 11:15 AM EDT","timestamp":1741706132},{"description":"Rendered: March 11, 2025 11:30 AM EDT","timestamp":1741707031},{"description":"Rendered: March 11, 2025 11:45 AM EDT","timestamp":1741707935},{"description":"Rendered: March 11, 2025 12:00 PM EDT","timestamp":1741708838},{"description":"Rendered: March 11, 2025 12:15 PM EDT","timestamp":1741709737},{"description":"Rendered: March 11, 2025 12:30 PM EDT","timestamp":1741710639},{"description":"Rendered: March 11, 2025 12:45 PM EDT","timestamp":1741711532},{"description":"Rendered: March 11, 2025 1:00 PM EDT","timestamp":1741712431},{"description":"Rendered: March 11, 2025 1:15 PM EDT","timestamp":1741713330},{"description":"Rendered: March 11, 2025 1:30 PM EDT","timestamp":1741714235},{"description":"Rendered: March 11, 2025 1:45 PM EDT","timestamp":1741715131},{"description":"Rendered: March 11, 2025 2:00 PM EDT","timestamp":1741716031},{"description":"Rendered: March 11, 2025 2:15 PM EDT","timestamp":1741716932},{"description":"Rendered: March 11, 2025 2:15 PM EDT","timestamp":1741716933},{"description":"Rendered: March 11, 2025 2:30 PM EDT","timestamp":1741717833},{"description":"Rendered: March 11, 2025 2:45 PM EDT","timestamp":1741718729},{"description":"Rendered: March 11, 2025 3:00 PM EDT","timestamp":1741719635},{"description":"Rendered: March 11, 2025 3:00 PM EDT","timestamp":1741719635},{"description":"Rendered: March 11, 2025 3:15 PM EDT","timestamp":1741720531},{"description":"Rendered: March 11, 2025 3:30 PM EDT","timestamp":1741721433},{"description":"Rendered: March 11, 2025 3:45 PM EDT","timestamp":1741722333},{"description":"Rendered: March 11, 2025 4:00 PM EDT","timestamp":1741723232},{"description":"Rendered: March 11, 2025 4:15 PM EDT","timestamp":1741724132},{"description":"Rendered: March 11, 2025 4:30 PM EDT","timestamp":1741725031},{"description":"Rendered: March 11, 2025 4:45 PM EDT","timestamp":1741725934},{"description":"Rendered: March 11, 2025 5:00 PM EDT","timestamp":1741726834},{"description":"Rendered: March 11, 2025 5:15 PM EDT","timestamp":1741727736},{"description":"Rendered: March 11, 2025 5:30 PM EDT","timestamp":1741728636},{"description":"Rendered: March 11, 2025 5:45 PM EDT","timestamp":1741729536},{"description":"Rendered: March 11, 2025 6:00 PM EDT","timestamp":1741730432},{"description":" [... cut ...] "Sunny. Not as cool with highs in the lower 60s.","Tuesday night":"Mostly clear in the evening, then becoming partly cloudy. Mild. Lows in the lower 40s.","Wednesday":"Mostly sunny. Warm with highs in the upper 60s.","Wednesday night":"Mostly cloudy. Unseasonably mild. Lows in the upper 40s.","Thursday":"Partly sunny. A chance of showers in the morning, then showers likely in the afternoon. Warm with highs in the mid 60s. Chance of rain 60 percent."},"forecastTime":"300 PM EDT Fri Mar 14 2025"}`

RyokoAsakura, 2025-03-15:
Things seem to have repaired themselves with the next hours run, but I still wonder what caused the error. Leaving ticket open for now.

JVital2013, 2025-04-09:
Hello!

Sorry for the late response. This bug appears to have happened in your AFM or PFM product file. The one used varies by location. What somewhat commonly happens is a space gets dropped in the matrix somewhere. I have some protections around this, but as you can see here, it can be difficult to work out what the value is supposed to be when this file gets malformed (especially if more than one space is dropped):

[image]

I'm not yet sure what causes spaces to get dropped like this. Both satdump and goestools produce files with a dropped space from time to time. I'm guessing it's an uplink error or a typo from an underpaid NWS intern 😄.

For now, the best solution is to set `debug=false` in config.ini unless you're actively debugging. Doing this should make it not display errors - but the errored information will be missing. In your case, humidity and clouds

RyokoAsakura, 2025-04-10:
Noted and will do. Thanks for clearing it up! I'm gonna close this as resolved. Thank you again!