JVital2013/vitality-goes #95: False Color images not showing.

pumatrax, 2025-04-09:
Hi,

I recently performed an update do to GOES-16 switching to GOES-19. After the update I could not see the False Color full disk images. I see there is a different in the filter you have to look for the full color image. In you abi.ini its set to ABI_False_Color but the images for full color satdump is saving is abi_rgb_GEO_False_Color.jpg.

When I change the filter to GEO_False_Color_map and GEO_False_Color I can now see the False Color images. 

[image]

JVital2013, 2025-04-09:
Hello! Thanks for reaching out.

It appears you are running an old version of SatDump, or you have overridden SatDump's configs. SatDump 1.2.0, and early nightlies of 1.2.1, named the composite `GEO_False_Color*`. SatDump 1.2.1 and newer renamed it to `ABI_False_Color*` as the composite is specific to GOES-R's ABI sensor.

To be specific, the change in SatDump happened [here](https://github.com/SatDump/SatDump/commit/8913dc1d618af4554dde6bae5ba32a280cd3a47c) on October 7, 2024, and Vitality GOES's sample configs were updated a week later in the [v1.5.3 release](https://github.com/JVital2013/vitality-goes/releases/tag/v1.5.3).

Your patch works, but it is worth updating SatDump because the current version now supports L2 product processing.

closed 2025-04-09 by JVital2013

pumatrax, 2025-04-10:
Hi, I updated SatDump and it's still saving the files as GEO. I'm running 1.2.3-34d67695. Where are SatDumps configs that specify it?  I did a git pull to the repo I installed it with. rebuilded and installed. 

[image]

pumatrax, 2025-04-10:
I removed the satdump program and rebuilt, the files are saving correctly now.