JVital2013/vitality-goes #96: Barometric Pressure and Weather?

pumatrax, 2025-04-11:
HI,

I am using SatDump to download from GOES19. I don't see Barometric Pressure in Vitality, at all. Is that expected? What should I be seeing for Weather? 

[image]

JVital2013, 2025-04-21:
Hello!

You're seeing N/A for "Weather" and "Barometric Pressure" because that's what is being transmitted in the Regional Weather Roundup files for Staten Island. See here for an example from about an hour ago:

```
NYZ071-072-176-178-NJZ106-104-211500-
New York City Metro Area

CITY           SKY/WX    TMP DP  RH WIND       PRES   REMARKS
Central Park   CLOUDY    54  23  30 SE9G20    30.36S
World Trd Ctr  NOT AVBL
Bronx Lehman C   N/A     52  32  46 E7          N/A
LaGuardia Arpt PTSUNNY   53  23  31 SE14      30.34S
Queens College   N/A     52  28  40 SE9         N/A
Kennedy Intl   CLOUDY    53  30  41 E17       30.35S
Breezy Point     N/A    N/A N/A N/A E10G16      N/A
Brooklyn Coll    N/A     50  32  50 E12         N/A
Staten Island    N/A     52  34  50 SE9         N/A
Newark/Liberty CLOUDY    56  28  34 SE10G18   30.34S
Teterboro      CLOUDY    56  27  32 SE13G22   30.33F
$$
```

As for why they're not transmitting that information in the regional weather roundup: I have no idea. You could ask them, and link them to this page so they can see what you're talking about: https://forecast.weather.gov/product.php?site=NWS&issuedby=okx&product=RWR.

pumatrax, 2025-04-22:
I switched to Central Park and can get data. That works for me. Thanks!