JVital2013/vitality-goes #97: Add a way to collapse or expand viewing images

pumatrax, 2025-04-11:
Can you add a way to expand or collapse all the images to save time opening them all, or closing if need be? 

[image]

JVital2013, 2025-04-21:
Until I get around to adding this, a workaround to do this is to add a new bookmarklet in whatever web browser you are using. In Chrome, right-click on your bookmarks bar and click "Add Page". Set the name to whatever you like, then set the url to the following:

```javascript
javascript:Array.from(document.getElementsByClassName("prettyBoxHeader")).forEach(element=>{element.click()})
```

[image]

Then, navigate to the Vitality GOES page you want to view, and click the bookmarklet. Technically this toggles the state of all cards, but it might be close enough until I can get this feature added.

pumatrax, 2025-04-22:
> JVital2013

Awesome! That works for now. Thank you.