JVital2013/vitality-goes #98: The Server Returned bad alert data:

Dentonknifeworks, 2025-04-20:
For the life of me I cant find the issue need some help.

[image]

This is my Location 
[location]
radarCode = REFUS
stateAbbr = TX
wxZone = TXZ103
orig = FWDTX
rwrOrig = TX-SJT
city = DENTON
lat = 33.00
lon = -97.98
timezone = America/Chicago

Thanks bet is just a simple thing I am overlooking.

nbkhwjm, 2025-04-20:
Whats your version of php?

Whats the value of error_reporting in the php ini...

(I would get this from a phpinfo() page so you know what ini is actually being used)

Dentonknifeworks, 2025-04-20:
"Whats the value of error_reporting in the php ini..."
error_reporting = E_ALL & ~E_DEPRECATED & ~E_STRICT

PHP version is 8.2.28 per phpinfo() page

nbkhwjm, 2025-04-20:
for what its worth, when I use your config in my setup it works fine..  I did notice if there are errant characters in the config.ini it errors out for me...

Dentonknifeworks, 2025-04-20:
> for what its worth, when I use your config in my setup it works fine.. I did notice if there are errant characters in the config.ini it errors out for me...

What version of PHP are you using?

nbkhwjm, 2025-04-20:
i'm using php version 7.4, but im not saying you need to drop back... ultimately if there was a change in php 8+ then this needs to be investigated.  I have another setup using php 8, I'll see how it acts with it.  

In the interim if debug is on in config.ini, turn it off and see how it works..  if any additional text is produced by dataHandler (Error or echo) it prevents it from working...  so the error statements may be benign, but are blowing up the datahandler parsing

Dentonknifeworks, 2025-04-20:
> i'm using php version 7.4, but im not saying you need to drop back... ultimately if there was a change in php 8+ then this needs to be investigated. I have another setup using php 8, I'll see how it [...]

Yes turning off the debug only shortens the Error MSG still an error

[image]

nbkhwjm, 2025-04-20:
Ok, so lets see what the server is returning to these... when the first page is loaded, these are called in the background to load data.  you can call these directly and lets see if there is an error in there.

http://YOUR_HOST/dataHandler.php?type=preload
http://YOUR_HOST/dataHandler.php?type=weatherJSON
http://YOUR_HOST/dataHandler.php?type=alertJSON

nbkhwjm, 2025-04-20:
if you make rwrOrig = FWDTX the same as Orig does it work?

Dentonknifeworks, 2025-04-20:
http://YOUR_HOST/dataHandler.php?type=alertJSON

I get this Page isnt working

nbkhwjm, 2025-04-20:
this is what I get..

`{
"spaceWeatherAlerts":[],
"weatherWarnings":[],
"hurricaneStatement":[],
"localEvacuations":[],
"civilDangerWarnings":[],
"amberAlerts":[],
"blueAlerts":[],
"localEmergencies":[]
}`

maybe re-enable debug and see if any messages..

Dentonknifeworks, 2025-04-21:
> this is what I get..
> 
> `{ "spaceWeatherAlerts":[], "weatherWarnings":[], "hurricaneStatement":[], "localEvacuations":[], "civilDangerWarnings":[], "amberAlerts":[], "blueAlerts":[], "localEmergencies":[] }`
> 
> maybe re-enable debug and see if any messages..

Enabling debug still blank page

Dentonknifeworks, 2025-04-21:
Correction on my part

Getting with Debug on 

Warning: Undefined array key "time" in /var/www/html/dataHandler.php on line 1015

Deprecated: substr_replace(): Passing null to parameter #1 ($string) of type array|string is deprecated in /var/www/html/dataHandler.php on line 1015

Warning: Undefined array key "date" in /var/www/html/dataHandler.php on line 1016

Warning: Undefined array key "timezone" in /var/www/html/dataHandler.php on line 1016

Deprecated: trim(): Passing null to parameter #1 ($string) of type string is deprecated in /var/www/html/dataHandler.php on line 1016

Fatal error: Uncaught Exception: DateTimeZone::__construct(): Unknown or bad timezone () in /var/www/html/dataHandler.php:1016 Stack trace: #0 /var/www/html/dataHandler.php(1016): DateTimeZone->__construct() #1 {main} thrown in /var/www/html/dataHandler.php on line 1016

JVital2013, 2025-04-21:
Hello @Dentonknifeworks!

Looking into the timezone stuff is a red herring. The problem has to do with how your special weather statements are formatted.

This is your correct config:

```ini
[location]
radarCode = REFUS
stateAbbr = TX
wxZone = TXZ103
orig = FWDTX
rwrOrig = SJTTX
city = DENTON
lat = 33.00
lon = -97.98
timezone = America/Chicago
```

However, Vitality GOES is failing to parse your special weather statements. I'll do a bit of digging and get back to you.

nbkhwjm, 2025-04-21:
Ahh ok, here is what I did for the record...

We did try to change the rwrOrig value and he stated it still failed.  Also I used his exact settings (with the wrong rwrOrig) and though the Current Weather did not appear I didn't see any errors, so focused on DateTimeZone.

I have seen these issues with the new PHP DateTimeZone, and it seems to deal with the lack of supported timezone identifiers and installation of tzdata_legacy fixed it in some cases.

Dentonknifeworks, 2025-04-21:
> Hello [@Dentonknifeworks](https://github.com/Dentonknifeworks)!
> 
> Looking into the timezone stuff is a red herring. The problem has to do with how your special weather statements are formatted.
> 
> This is your correct config:
> 
> [location]
> radarCode = REFUS
> stateAbbr = TX
> wxZone = TXZ103
> orig = FWDTX
> rwrOrig = SJTTX
> city = DENTON
> lat = 33.00
> lon = -97.98
> timezone = America/Chicago
> However, Vitality GOES is failing to parse your special weather statements. I'll do a bit of digging and get back to you.

Thanks for the info. I changed the config to the correct one you listed. Same issue but nice to know that the config is now correct. I will wait for your suggestion.

JVital2013, 2025-04-21:
Could either of you add the following lines into dataHandler.php at line 992? It will dump out the files causing the problem:

[image]

Afterwords, give me the error displayed in current weather, then delete these debug lines. Thanks!

nbkhwjm, 2025-04-21:
Mine doesn't throw the error like his, but here are the results of the var_dump when I use his config

`The server returned bad alert data:

array(7) { 
[9046]=> string(74) "/goes/emwin/A_WWUS84KFWD190231_C_KWIN_20250419023113_244565-1-SPSFWDTX.TXT" [27150]=> string(74) "/goes/emwin/A_WWUS84KFWD191352_C_KWIN_20250419135242_280922-1-SPSFWDTX.TXT" [30309]=> string(74) "/goes/emwin/A_WWUS84KFWD192251_C_KWIN_20250419225143_308785-1-SPSFWDTX.TXT" [33817]=> string(74) "/goes/emwin/A_WWUS84KFWD190930_C_KWIN_20250419093038_267698-1-SPSFWDTX.TXT" [46829]=> string(74) "/goes/emwin/A_WWUS84KFWD191328_C_KWIN_20250419132833_279956-1-SPSFWDTX.TXT" [70801]=> string(74) "/goes/emwin/A_WWUS84KFWD200752_C_KWIN_20250420075233_338170-1-SPSFWDTX.TXT" [79300]=> string(74) "/goes/emwin/A_WWUS84KFWD192244_C_KWIN_20250419224437_308383-1-SPSFWDTX.TXT"
 }`

Dentonknifeworks, 2025-04-21:
The server returned bad alert data:

array(34) { [23076]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD041917_C_KWIN_20250404191801_167187-1-SPSFWDTX.txt" [51963]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD030749_C_KWIN_20250403074925_046305-1-SPSFWDTX.txt" [66860]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD031229_C_KWIN_20250403122940_064752-1-SPSFWDTX.txt" [75718]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD021841_C_KWIN_20250402184153_995067-1-SPSFWDTX.txt" [87636]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD021928_C_KWIN_20250402192855_998076-1-SPSFWDTX.txt" [90505]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD150912_C_KWIN_20250415091256_969410-1-SPSFWDTX.txt" [95476]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD192244_C_KWIN_20250419224437_308383-1-SPSFWDTX.txt" [99292]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD041920_C_KWIN_20250404192101_167297-1-SPSFWDTX.txt" [137478]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD021234_C_KWIN_20250402123416_974370-1-SPSFWDTX.txt" [141688]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD021631_C_KWIN_20250402163115_987750-1-SPSFWDTX.txt" [150043]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD191352_C_KWIN_20250419135242_280922-1-SPSFWDTX.txt" [157231]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD030951_C_KWIN_20250403095119_055245-1-SPSFWDTX.txt" [163282]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD041840_C_KWIN_20250404184044_164875-1-SPSFWDTX.txt" [168905]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD200752_C_KWIN_20250420075233_338170-1-SPSFWDTX.txt" [209926]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD150840_C_KWIN_20250415084056_967272-1-SPSFWDTX.txt" [228303]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD031840_C_KWIN_20250403184054_084720-1-SPSFWDTX.txt" [240537]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD021811_C_KWIN_20250402181204_993683-1-SPSFWDTX.txt" [262518]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD042136_C_KWIN_20250404213644_175944-1-SPSFWDTX.txt" [291185]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD190930_C_KWIN_20250419093038_267698-1-SPSFWDTX.txt" [291933]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD040933_C_KWIN_20250404093349_135459-1-SPSFWDTX.txt" [293360]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD192251_C_KWIN_20250419225143_308785-1-SPSFWDTX.txt" [303795]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD040724_C_KWIN_20250404072506_126742-1-SPSFWDTX.txt" [305053]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD151330_C_KWIN_20250415133010_982096-1-SPSFWDTX.txt" [339255]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD021809_C_KWIN_20250402180949_993372-1-SPSFWDTX.txt" [356233]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD031026_C_KWIN_20250403102657_057489-1-SPSFWDTX.txt" [366008]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD190231_C_KWIN_20250419023113_244565-1-SPSFWDTX.txt" [367249]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD030937_C_KWIN_20250403093753_054408-1-SPSFWDTX.txt" [370408]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD030944_C_KWIN_20250403094435_054796-1-SPSFWDTX.txt" [375978]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD020029_C_KWIN_20250402003008_932428-1-SPSFWDTX.txt" [449064]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD022015_C_KWIN_20250402201601_001511-1-SPSFWDTX.txt" [453247]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD191328_C_KWIN_20250419132833_279956-1-SPSFWDTX.txt" [473827]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD041848_C_KWIN_20250404184901_165344-1-SPSFWDTX.txt" [486858]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD020058_C_KWIN_20250402005900_933916-1-SPSFWDTX.txt" [498581]=> string(77) "/home/pi/EMWIN/A_WWUS84KFWD021745_C_KWIN_20250402174622_992166-1-SPSFWDTX.txt" }

JVital2013, 2025-04-21:
I found it - the problem is being caused by A_WWUS84KFWD151330_C_KWIN_20250415133010_982096-1-SPSFWDTX.txt, which is issued by a different office. I'll see if I can update the code to handle this condition.

JVital2013, 2025-04-21:
As a note, it is worth configuring the Cleanup-EmwinText.sh scripts. You're accruing old data, which is slowing things down over time.

Dentonknifeworks, 2025-04-21:
> As a note, it is worth configuring the Cleanup-EmwinText.sh scripts. You're accruing old data, which is slowing things down over time.

I will wait for your changes before cleaning them so that we can test.

JVital2013, 2025-04-21:
OK, the bug should now be fixed. You can download update dataHandler.php from the copy [here](https://github.com/JVital2013/vitality-goes/blob/main/html/dataHandler.php).

Effectively the problem was that some files have a header that look like this, with an extra "Issued by" line:

```
Special Weather Statement
National Weather Service Fort Worth TX
Issued by National Weather Service Shreveport LA
830 AM CDT Tue Apr 15 2025
```

While parsing files like this for forecast data works fine, it did not work fine for alert data because it was parsed differently. The updated code in 5907d39 should fix it.

Dentonknifeworks, 2025-04-21:
Yes that has fixed it.
Thanks
Closing the issue.

closed 2025-04-21 by Dentonknifeworks