JVital2013/vitality-goes #101: New Installation after crash

njrusty, 2025-05-11:
I had replace my ssd on my RP4 and reinstall everything (was running great for a year).

Thought I had followed directions step by step but missing something now. After installation of all software, updates and Apache. When I enter my goes ip address, I get the following. Any help would be appreciated. Thanks!

[image]

JVital2013, 2025-05-13:
Hello!

It looks like the Vitality GOES web files are missing from /var/www/html (or, Apache got configured to serve files from another path, which is less likely). Assuming it's just the files missing:

1. cd to the git directory of Vitality GOES
2. `cp -r html /var/www/html`

njrusty, 2025-05-13:
I got the following error:

[image]

JVital2013, 2025-06-02:
Sorry, I was unclear. You need to do it one directory up (in your case, ~/vitality-goes, not ~/vitality-goes/.git)